paviro/MMM-PIR-Sensor #1: Uncaught Exception when adding module

PtrBld, 2016-05-08:
Hi,
I get an uncaught exception when I add your module to the config. The PIR sensor is connected correctly (tested with script) and works well when I use the `node serveronly` - way of starting and browse it with a browser but when using electron and pm2 it gives me the following error:
`mm-0 (out): Whoops! There was an uncaught `exception...`

Seems like there are permissions missing?! 
I followed your guide to install the module, so can you say what I am missing?

paviro, 2016-05-08:
What exactly is the `uncaught exception`? Does not seem like a problem with the module. Have you added your user to the `gpio` group and rebooted?

PtrBld, 2016-05-08:
Yes. Like I said it works if I use the browser version but stops working if I use electron...
I am starting to think that it is a electron problem anyway since I investigated further and pointed out that I get a version error like `mm-0 (err): App threw an error when running [Error: Module version mismatch. Expected 47, got 48.] beforehand.

Do you have any idea why this could be happening? Otherwise I think I should open an issue in the main repository.
`

paviro, 2016-05-08:
Seems like your node version is too new! Are you using node 6? Some modules are not yet compatible with it (I had the same problem) try downgrading to the latest node 5 release and the execute `npm install` again.

closed 2016-05-08 by paviro