paviro/MMM-PIR-Sensor #2: After Installation, version error

KilleKat, 2016-05-15:
Hey,

I want to use this module. After I go through your installation steps. I only see a black screen on the mirror and nothing happens. Also the console show a version error (see screenshot)

Unfortunatelly I am not a NodeJS or npm expert,... so maybe i have done something wrong.
[image]

paviro, 2016-05-15:
Please use Node v5 not v6 since it is not compatible yet.

KilleKat, 2016-05-15:
How I do this? How can i downgrade?

paviro, 2016-05-15:
http://stackoverflow.com/questions/7718313/how-to-change-to-an-older-version-of-node-js :)

KilleKat, 2016-05-15:
I have downgrade, but still get this issue :(
[image]

paviro, 2016-05-15:
Run npm install again.

KilleKat, 2016-05-15:
In which folder I have to run 'sudo npm install' ? I have done it in the 'MMM-PIR-Sensor' and in the 'MagicMirror' folder but I still get the verison issue

paviro, 2016-05-15:
What das node -v say?

KilleKat, 2016-05-15:
pi@raspberrypi:~ $ node -v
v5.11.1
pi@raspberrypi:~ $

paviro, 2016-05-15:
try removing the module folder and loading it from git again.

KilleKat, 2016-05-15:
Now it s working fine!!! Yeah, thx! Can you maybe send me a picture how you have setup your PIR sensor?

closed 2016-05-15 by KilleKat