paviro/MMM-PIR-Sensor #7: Turning off my monitor or nothing to show

murattyilmaz, 2016-07-21:
Hello, Thanks for this cool module. I think it should be a default module inside.

But I'm facing a problem. When I activate this module, app starts as expected then 5 sec later my monitor is turning off and says me "no signal" and I think this is expected. Then I moving in front of the sensor and monitor say nothing and doesn't show anything. I'm not sure what is wrong and no idea how can fix or find the  issue. Need help pls.

paviro, 2016-07-27:
Seems like the sensor does not work correctly :/ Make sure you have set up the correct pins!

murattyilmaz, 2016-08-01:
I just checked it out via an small python app and seems like there is no problem with the sensor.
How can I debug your module? I'm not sure to module can live my monitor or now.

fixing-it, 2016-08-21:
Same here, monitor is switching on but screen stays black.
Entering `/opt/vc/bin/tvservice --preferred && sudo chvt 6 && sudo chvt 7` via ssh everything shows up as expected until it gets turned off again.

paviro, 2016-09-11:
Since '/opt/vc/bin/tvservice --preferred && sudo chvt 6 && sudo chvt 7' is in the module now it should work :)

closed 2016-09-11 by paviro

cyberdie, 2017-05-30:
Same here with last version.
Monitor switches on but screen doesn't show any module.
Disabling module in config.js all modules shows as expected.
Any clue?