paviro/MMM-PIR-Sensor #9: Error: Module version mismatch. Expected 49, got 48.

JMishou, 2016-08-02:
App threw an error during load
Error: Module version mismatch. Expected 49, got 48.
    at Error (native)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:167:20)
    at Object.Module._extensions..node (module.js:568:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:167:20)
    at Module.load (module.js:458:32)
    at tryModuleLoad (module.js:417:12)
    at Function.Module._load (module.js:409:3)
    at Module.require (module.js:468:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/wiring-pi/lib/exports.js:1:97)
Error: Module version mismatch. Expected 49, got 48.
    at Error (native)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:167:20)
    at Object.Module._extensions..node (module.js:568:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:167:20)
    at Module.load (module.js:458:32)
    at tryModuleLoad (module.js:417:12)
    at Function.Module._load (module.js:409:3)
    at Module.require (module.js:468:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/wiring-pi/lib/exports.js:1:97)

Unless I am reading this wrong this is expecting a Node module version that is not released yet?

The current module release is 48 https://nodejs.org/en/download/releases/

 I have tried node version v6.3.0, 6.3.1, and 5.12.0.  

I have followed all of your suggestions here even though that a downgrade to a previous version and not an upgrade to an unreleased version..  https://github.com/paviro/MMM-PIR-Sensor/issues/2

I have tried reinstalling electron, rebuilding it, removing the module folder and re cloning it.  I'm at a loss.

The only other instance of this I could find is here https://github.com/don/node-eddystone-beacon/issues/40  Electron-rebuild seemed to work in their case.  Not so much for me.

Got any suggestions?  Much appreciated.

-Jason.

paviro, 2016-08-02:
> at Object. (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/wiring-pi/lib/exports.js:1:97)
> The only other instance of this I could find is here don/node-eddystone-beacon#40 <https://github.com/don/node-eddystone-beacon/issues/40> Electron-rebuild seemed to work in their case. Not so much for me.

This error normally comes when you run npn install inside the module folder on another node version you are using now to run the script. Try running npm install again in the module directory.

JMishou, 2016-08-03:
npm install didn't work.  So I stared over with a clean raspian image and still after installing MM and PIR I get the errors.  

Please let me know if there is anything you would like me to try.

mm-0 Whoops! There was an uncaught exception...
mm-0 Error: Module version mismatch. Expected 49, got 48.
mm-0     at Error (native)
mm-0     at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:167:20)
mm-0     at Object.Module._extensions..node (module.js:568:18)
mm-0     at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:167:20)
mm-0     at Module.load (module.js:458:32)
mm-0     at tryModuleLoad (module.js:417:12)
mm-0     at Function.Module._load (module.js:409:3)
mm-0     at Module.require (module.js:468:17)
mm-0     at require (internal/module.js:20:19)
mm-0     at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/wiring-pi/lib/exports.js:1:97)

paviro, 2016-08-03:
> mm-0 at Object. (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/wiring-pi/lib/exports.js:1:97)

It's probably best to wait until I have the time to remove wiringpi like I want to do in the next version :)

JMishou, 2016-08-03:
Where is the source of the wiring pi libraries that you are using?  There seem to be several node.js wrappers.

paviro, 2016-08-03:
Check the projects json file. The name is in the dependencies there :)

Bangee44, 2016-08-04:
I've got the same error. How can I fix that?

JMishou, 2016-08-09:
I don't have much experience with node so I cannot tell you how to fix this but I do have a workaround for this specific application.

I created a daemon in C that you can run at startup that will turn the monitor on and off.  It also uses wiring pi and is essentially the same exact code only written in C.

https://github.com/JMishou/piMotionSensor

I'm done with this issue paviro... it still exists but I don't care to try and figure it out anymore.  My gut is it is not with your code but the wiringpi library.

If you would like me to close it I will.

gylian, 2016-08-09:
I also have this error. Tried a clean install. Raspbian Jessie, updated node to 6.3.1, installed Magicmirror then this module and got this error.

fsuk, 2016-09-10:
After running npm install in the MMM-PIR-Sensor directory run the following command:
`
npm rebuild --runtime=electron --target=1.3.4 --disturl=https://atom.io/download/atom-shell --abi=49
`
Credit to  TomAshley303  here https://github.com/electron-userland/electron-builder/issues/453

paviro, 2016-09-11:
This should now be resolved, I installed it completely normal without any problems.

closed 2016-09-11 by paviro

frekel, 2017-01-02:
Please reopen this issue. 
None of the suggestions work for me:

npm rebuild
adding the user to the group
rebooting
clean install

mlevy94, 2017-04-03:
Issue is present again with the latest update to MM and node. Expecting version 53, getting version 49.

jwu910, 2017-04-03:
I managed to get this resolved last night on my machine. I can't say for sure what did it, but i want to say it was following the lines of:
-npm uninstall electron
-npm rebuild/install electron (??) It was late last night and I wasn't expecting it to work.

mlevy94, 2017-04-05:
I'm still getting the same error even after the rebuild.

toniweser, 2017-04-30:
I got the same problem (Expect 50, got 53). I tried a bunch of stuff.

Finally the solution from jwu910 worked!

Execute the following commands in /pi/magicmirror/ (your magicmirror installation folder):

`npm uninstall electron`
`npm rebuild`
`npm install electron`

My System:
Electron version 1.6.6
Node JS version 7.9
MagicMirror version 2.1.1

semox, 2017-08-10:
This does not work on my side. Are there any workarounds?

semox, 2017-08-11:
This did the trick for me:
`cd ~/MagicMirror`
`git pull`
`rm -rf node_modules/`
`npm cache clean`
`npm install`
`for f in $(find /home/pi/MagicMirror/modules -maxdepth 1 -name "MMM-*" -type d); do cd $f; git pull && npm install; done`

jwu910, 2017-08-11:
Interesting fix @semox, Like a purge and reinstall. No clue why it wouldn't be working but most of the solutions people have found seem to be revolving around uninstalling node module(s) and reinstalling them.