paviro/MMM-PIR-Sensor #12: ReferenceError: pir is not defined

kexroman, 2016-09-07:
Hi, I got the error when try to use this module :

> **pi@raspberrypi:~/.pm2/logs $ tail -25   mm-error-0.log**
> Missing error handler on `socket`.
> ReferenceError: pir is not defined
>     at Class.socketNotificationReceived (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_helper.js:54:7)
>     at Socket.<anonymous> (/home/pi/MagicMirror/modules/node_modules/node_helper/index.js:98:10)
>     at emitTwo (events.js:106:13)
>     at Socket.emit (events.js:191:7)
>     at Socket.onevent (/home/pi/MagicMirror/node_modules/socket.io/lib/socket.js:335:8)
>     at Socket.socket.onevent (/home/pi/MagicMirror/modules/node_modules/node_helper/index.js:91:13)
>     at Socket.onpacket (/home/pi/MagicMirror/node_modules/socket.io/lib/socket.js:295:12)
>     at Client.ondecoded (/home/pi/MagicMirror/node_modules/socket.io/lib/client.js:193:14)
>     at Decoder.Emitter.emit (/home/pi/MagicMirror/node_modules/component-emitter/index.js:134:20)
>     at Decoder.add (/home/pi/MagicMirror/node_modules/socket.io-parser/index.js:247:12)
> **pi@raspberrypi:~/MagicMirror/modules/MMM-PIR-Sensor $** grep -r pir *
> node_helper.js:      this.pir = new Gpio(this.config.sensorPIN, 'in', 'both');
> node_helper.js:      pir.watch(function(err, value) {
> pi@raspberrypi:~/MagicMirror/modules/MMM-PIR-Sensor $
> 

What I have to do with this variable?  
Thanks,

shrimp69, 2016-09-07:
Same happens to me

paviro, 2016-09-07:
Will setup a mirror on the weekend and fix it then :)

fsuk, 2016-09-10:
It looks like the variable pir line in node_helper is not declared. Not being familiar enough with how the modules are developed and where the variables are declared I can only offer a workaround:

In node_helper.js change:
`//Setup pins
this.pir = new Gpio(this.config.sensorPIN, 'in', 'both');`
to
`//Setup pins
var pir = new Gpio(this.config.sensorPIN, 'in', 'both');`

paviro, 2016-09-11:
Should work now :)

closed 2016-09-11 by paviro

kexroman, 2016-09-12:
Yep, super, it works.

commit a0646f9f (2016-09-11): Fixed #12 pir is not defined.