paviro/MMM-PIR-Sensor #16: MMM-PIR-Sensor is not working

XxXCobraXxX, 2017-01-08:
**Hy Paviro,**

i have a problem with your module.

My PIR Sensor is working -> i have tested it. I have connected it to GPIO-Pin 7 (GPIO4).

This is my config from MMM-PIR-Sensor in MagicMirror config file:

> {
	module: 'MMM-PIR-Sensor',
	classes: 'default jeder',
	config: {
		// See 'Configuration options' for more information.
		sensorPIN: 4
	}
},

And this are the ERRORS from pm2 logs:

> 0|mm       | { Error: EBUSY: resource busy or locked, write
> 0|mm       |     at Error (native)
> 0|mm       |     at Object.fs.writeSync (fs.js:786:20)
> 0|mm       |     at Object.fs.writeFileSync (fs.js:1353:24)
> 0|mm       |     at new Gpio (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/onoff/onoff.js:88:8)
> 0|mm       |     at Class.socketNotificationReceived (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_helper.js:47:18)
> 0|mm       |     at Socket.<anonymous> (/home/pi/MagicMirror/modules/node_modules/node_helper/index.js:98:10)
> 0|mm       |     at emitTwo (events.js:106:13)
> 0|mm       |     at Socket.emit (events.js:191:7)
> 0|mm       |     at /home/pi/MagicMirror/node_modules/socket.io/lib/socket.js:503:12
> 0|mm       |     at _combinedTickCallback (internal/process/next_tick.js:67:7) errno: -16, code: 'EBUSY', syscall: 'write' }

I  don't know what i doing wrong and i have searched many hours in the web but without any positive solution. You are my last chance. Please help me.

Thanks in advance

paviro, 2017-01-10:
Any chance this is related? https://github.com/fivdi/onoff/issues/45

closed 2017-04-02 by paviro

M1dn8, 2017-04-05:
Hey everybody,

I really don't know what I did wrong!
I followed exactly your instructions and connected my PIR to GPIO 22(BCM) but nothing happened.

gpio read 22 = 1 --> so it's working but nothing happend 🤔

Could u imagine what my mistake is?

Thx for ur help and thx for your great work 👍🏼

Kolk123, 2017-07-29:
if got the same problem. how did you fixed it