paviro/MMM-PIR-Sensor #17: module not working PLEASE HELP

sainiaj, 2017-01-11:
**Hi this module use to work very well before.I just reinstalled the magic mirror and it is not working.
PLEASE HELP** 

**git clone https://github.com/paviro/MMM-PIR-Sensor.git
pi@raspberrypi:~/MagicMirror/modules $ cd MMM-PIR-Sensor
pi@raspberrypi:~/MagicMirror/modules/MMM-PIR-Sensor $ npm install**
npm WARN prefer global node-gyp@3.5.0 should be installed with -g

> epoll@0.1.20 install /home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build'

> Magic-Mirror-Module-PIR-Sensor@1.1.0 postinstall /home/pi/MagicMirror/modules/MMM-PIR-Sensor
> electron-rebuild -e ../../node_modules/electron-prebuilt

An unhandled error occurred inside electron-rebuild
Unable to find electron-prebuilt's version number, either install it or specify an explicit version

Error: Unable to find electron-prebuilt's version number, either install it or specify an explicit version
    at _callee$ (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/cli.js:74:17)
    at tryCatch (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/regenerator-runtime/runtime.js:63:40)
    at GeneratorFunctionPrototype.invoke [as _invoke] (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/regenerator-runtime/runtime.js:337:22)
    at GeneratorFunctionPrototype.prototype.(anonymous function) [as next] (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/regenerator-runtime/runtime.js:96:21)
    at GeneratorFunctionPrototype.tryCatcher (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bluebird/js/release/util.js:16:23)
    at PromiseSpawn._promiseFulfilled (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bluebird/js/release/generators.js:97:49)
    at /home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bluebird/js/release/generators.js:201:15
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/cli.js:197:4)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)

npm ERR! Linux 4.4.34-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! node v6.9.4
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! Magic-Mirror-Module-PIR-Sensor@1.1.0 postinstall: `electron-rebuild -e ../../node_modules/electron-prebuilt`
npm ERR! Exit status 255
npm ERR!
npm ERR! Failed at the Magic-Mirror-Module-PIR-Sensor@1.1.0 postinstall script 'electron-rebuild -e ../../node_modules/electron-prebuilt'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the Magic-Mirror-Module-PIR-Sensor package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron-rebuild -e ../../node_modules/electron-prebuilt
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs Magic-Mirror-Module-PIR-Sensor
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls Magic-Mirror-Module-PIR-Sensor
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/modules/MMM-PIR-Sensor/npm-debug.log

-------------------------------------------

[npm-debug.zip](https://github.com/paviro/MMM-PIR-Sensor/files/698024/npm-debug.zip)

darkeagl, 2017-01-11:
I have the exact same error.
MagicMirror 2.1.0
Raspberry Pi B 3 Jessie with PIXEL

sainiaj, 2017-01-11:
try this mine is working now 

 cd ~/MagicMirror/modules/MMM-PIR-Sensor
 npm list
 npm rebuild --runtime=electron --target=1.4.3 --disturl=https://atom.io/download/atom-shell --abi=50
cd
 pm2 restart mm 

sudo reboot (just to make sure everything works)

darkeagl, 2017-01-11:
Wow thank you. That work liked a charm

sainiaj, 2017-01-12:
@darkeagl 
is it working for you, 
for me magic mirror is loading and other modules are working 
but the screen is not turning off !

UGHHH !

qistoph, 2017-01-13:
I've tried the commands suggested by @sainiaj. It compiles, but when starting I get errors:

````
Error: Module version mismatch. Expected 48, got 50.
    at Error (native)
    at Object.Module._extensions..node (module.js:597:18)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at bindings (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/epoll.js:1:99)
    at Module._compile (module.js:570:32)
````

Changing the `--abi=50` to `--abi=48` did not solve this issue. Any suggestions?

Running on Raspberry Pi 3 with Raspbian.

Atheose, 2017-01-14:
I too am getting the "fails on electron-rebuild -e" error. Any luck?

sainiaj, 2017-01-15:
I hope someone finds the fix soon, this is very annoying,
i used older version of rasbian jessie dec 2016 because it was working before,
but i think either magic mirror has been updated or PIR module causing this error

I hope someone finds a solution soon

Ps. does anyone know any third party program which can turn on/off monitor when motion detected

qistoph, 2017-01-15:
You could try this standalone python script:
https://github.com/deg0nz/MagicMirror-Display-Controller

sainiaj, 2017-01-15:
@qistoph let me try this one 
i tried this one it is working if i do the python script (manually) but not turning on and executing after reboot automatically  
www./2014/01/2/building-photo-frame-raspberry-pi-motion-detector/

Atheose, 2017-01-15:
Funny, I created my own little python script while waiting to do the same thing. It utilizes the "tvservice" command to turn the screen on and off:

```
from gpiozero import MotionSensor
import time
import os

pir = MotionSensor(4)
while True:
    if pir.motion_detected:
        os.system("tvservice -p")
        time.sleep(60)
        os.system("tvservice -o")
    time.sleep(0.5)
```

The script checks every half-second, and if motion is detected it turns on the screen and leaves it up for 60 seconds. At the end of the 60 seconds it turns off again.

qistoph, 2017-01-18:
Ok, did some research, this is what I've found.
The Pi3 running Jessie uses https://deb.nodesource.com/node_6.x (/etc/apt/sources.list.d/nodesource.list). This currently installs Node.js v6.9.4.
Node.js v6.9.1 uses (see "Get Node.js ABI" below).
The MagicMirror dependency `"electron": "^1.4.7"` currently installs electron v1.4.14, which uses ABI 50 (see "Get Electron ABI" below).
Installing the MMM-PIR-Sensor dependency `"onoff": "latest"` installs epoll which is compiled for Node.js during install, using ABI=48.
Starting MagicMirror with 'node serveronly' uses Node.js with ABI 48, which works.
When using electron with ABI 50, the ABI mismatches. "Module version mismatch. Expected 50, got 48."

Get Node.js ABI:
Check [Node.js releases](https://nodejs.org/en/download/releases/)
or run `echo "console.log(process.versions.modules);" | node`

Get Electron ABI:
`echo "console.log(process.versions.modules);" | ELECTRON_RUN_AS_NODE=1 ./node_modules/electron/cli.js`

**Solution**
Rebuild for electron (this should make modules use ABI 50):
`npm rebuild --runtime=electron --disturl=https://atom.io/download/atom-shell`

Start MagicMirror with electron, with either:
- `npm start`
- `pm2 start mm`
- `ELECTRON_RUN_AS_NODE=1 ./node_modules/electron/cli.js serveronly` (instead of `node serveronly`)

Jopyth, 2017-01-24:
Oh I wish i discovered this earlier. The main project switched from `electron-prebuilt` to the `electron` package, so this needs to be adapted here, too.

closed 2017-02-04 by paviro

qistoph, 2017-09-11:
This issue is suddenly back with my installation.

Reinstalling the module fails:

```
pi@magicmirror:~/MagicMirror/modules/MMM-PIR-Sensor $ npm install

> epoll@0.1.22 install /home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build'

> Magic-Mirror-Module-PIR-Sensor@1.1.0 postinstall /home/pi/MagicMirror/modules/MMM-PIR-Sensor
> electron-rebuild -e ../../node_modules/electron

sh: 1: electron-rebuild: not found

npm ERR! Linux 4.4.38-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! node v6.10.0
npm ERR! npm  v3.10.10
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! Magic-Mirror-Module-PIR-Sensor@1.1.0 postinstall: `electron-rebuild -e ../../node_modules/electron`
npm ERR! spawn ENOENT
npm ERR!
npm ERR! Failed at the Magic-Mirror-Module-PIR-Sensor@1.1.0 postinstall script 'electron-rebuild -e ../../node_modules/electron'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the Magic-Mirror-Module-PIR-Sensor package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron-rebuild -e ../../node_modules/electron
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs Magic-Mirror-Module-PIR-Sensor
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls Magic-Mirror-Module-PIR-Sensor
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/modules/MMM-PIR-Sensor/npm-debug.log
```

Anyone else having these issues too?

PR #18 (2017-02-04): Update to electron package
Need to check this at home, I think I fixed #17 locally a long time ago, when I switched to the develop branch at `2.1.0`. Not sure this works out-of-the-box, so do not merge before I (or someone else) tested it. There might be something missing.

Also it will break anyone's installation, who did not upgrade to MM `2.1.0` yet.