paviro/MMM-PIR-Sensor #21: Not working as expected

ssynix, 2017-02-18:
The behavior that I see with this module is my monitor constantly flashing on and off when the sensor detects any movement, and instantly turns off when no motion is detected.

I modified the source code to include a 20 sec timer. If there's any motion detected, the timer is reset. The monitor turns off once the timer goes off. This resulted in the kind of behavior I was expecting.

I'm wondering if there's something wrong with my setup or sensor which is resulting in the former behavior.

Strychy, 2017-02-22:
Hey ssynix would you be willing to share the modifications you have made to the code? I'm also experiencing this issue.

Patrick-Remy, 2017-02-28:
For additional software-side delay see my PR #20

KX900, 2017-03-01:
@ssynix 
could you share, how to add the timer?

ssynix, 2017-03-02:
Sorry, I was busy with work. I put it up on my fork: https://github.com/ssynix/MMM-PIR-Sensor

paviro, 2017-04-02:
Will probably be merged soon :)

closed 2017-04-02 by paviro