paviro/MMM-PIR-Sensor #23: Question not Issue - Behaviour when HDMI output is stopped

sorford638, 2017-03-02:
Hi,

My apologies if I am posting this in the wrong place but I am about to buy a second hand BenQ monitor (http://www.benq.co.uk/product/monitor/g2420hdbl) with DVI-D input and I wanted to check a couple of things.

1)  I am using a convertor (https://www.amazon.es/gp/product/B00JM9NHAU/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1) to switch from the HDMI ouput of the raspberry Pi to the DVI-D input of the BenQ g2420hdbl monitor.  Can you see this causing a problem?

2)  When the HDMI Output from the Raspberry Pi is stopped (but the cable still attached) will the monitor enter sleep mode or display the "No Signal" message?

Once again, my apologies if this is not the right place for this sort of query.

Al the best and thanks for making the effort to share your work,
Steve

Patrick-Remy, 2017-03-02:
I am using this plugin with a similar config, and it works very well! The monitor will turn off, even with adapter.

sorford638, 2017-03-02:
Thanks Patrick.  When you say "turn off" do you mean "off" or sleep mode?

Patrick-Remy, 2017-03-03:
Sleep mode. This behavior isn't different to a setup without adapter. 
Really turning off can be achieved with a Relais.  But then you have the problem that starting up the monitor will take some time. Sleep mode is enough for me.

paviro, 2017-04-02:
Please use the forum!

closed 2017-04-02 by paviro