paviro/MMM-PIR-Sensor #25: [Enhancement] Replace tvservice with vcgencmd

anmith, 2017-03-15:
Hi,
as tvservice always needs the virtual console thing, how about replacing it with vcgencmd?

sudo /opt/vc/bin/vcgencmd display_power 0

turns off display power while

sudo /opt/vc/bin/vcgencmd display_power 1

turns the power back again.

Detailed info: http://www.elinux.org/RPI_vcgencmd_usage

eckonator, 2018-04-05:
Hi @anmith,
i thing you're right. Here is my working vcgencmd-Replacment:

[https://pastebin.com/Znp2VY5b](url)

Just copy&paste this into this file:
`/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_helper.js`
and replace the whole existing code there.

Maybe it helps someone and hope it finds a way into the master branch of this module?

Simon089, 2018-04-09:
Hi,

I wanted to create a PR with some changes I made for resetting the timer when someone is still standing in front of the mirror. I will include the `vcgencmd`, too. 
So hopefully this finds its way to into the master branch.

ubertao, 2018-06-28:
PR #42 works great for me.

chvt refuses run without sudo.