paviro/MMM-PIR-Sensor #26: Question - Will Monitor Power Off or just show "No HDMI Connected"?

reillychase, 2017-04-07:
Hi I am working on building my magic mirror, and I think this plugin is the only module able to power off the display when no one is nearby, correct?

With my particular monitor, its settings are locked and the button to unlock them is broken, so no way to change it to power off.

If this module shuts down HDMI connection from the rpi to my monitor, the monitor will display "No HDMI Connected" screen, does that sound right? Any idea to work around this? Thanks

paviro, 2017-04-08:
You can use switch the power of by using a relay :)

reillychase, 2017-04-08:
Good idea, thanks for the suggestion! Messing with electricity scares me a bit, I think I'm going to actually get a Belkin Wemo switch and then I am hoping I can figure out how to modify your script to call a python script which will then turn the monitor power on/off when motion detected or not detected for X amount of time. I will also probably want to modify your script to be able to disable motion sensor sometimes like at night because the mirror is in my bedroom and I dont want it turning on when I wake up in the middle of the night or something ha

reillychase, 2017-04-09:
Update, I tested my monitor with the '/opt/vc/bin/tvservice' on/off commands and it did work actually! Even though the monitor is old / 2011. So thats great, no need for belkin now

tjinks17, 2017-05-31:
reillychase - I'd like to make the same change to the code as you that you posted on Apr 8. I'm a hack coder though ... What code exactly did you add to which program?  Thank you!