paviro/MMM-PIR-Sensor #27: Monitor repeatedly turning back on after "HDMI OFF" status

nyxxjoep, 2017-04-09:
Hi - I downloaded the latest build. My PIR sensor works great to wake up from the HDMI OFF state on my LCD. However, even if I'm away from the IR, the monitor indefinitely loops back from HDMI OFF to being awake. I have my timer set to 15 seconds - I wonder if that's too long and causing a conflict with my LCD hardware?

CriticalPoint, 2017-05-08:
I had the same problem with my PIR behind the mirror film/glass. I rectified by using a different PIR above the mirror (as the previous was fixed in place), perhaps either a faulty PIR, or that the sudden change in light from the LCD panel activated the PIR, switching the screen back on, causing a loop. Try isolating it and testing that way..?

renaudjc, 2017-05-17:
Hi Guy ! Seems I experience  the same issue, monitor repeats "power saving mode" trying continously to wake up after a detection. Did you fix it ?

Corburn84, 2017-12-09:
Having the same problem. Monitor switches off and on every second... without having the chance to show a picture. Does anyone have a hint how to fix that?

Skorpionbird, 2018-12-25:
My PIR is not behind the Mirrorglas. 
Der PIR-Sensor funktioniert, wenn ich mich innerhalb von 10 Sekunden bewege, wenn der HDMI-Ausgang des Pi deaktiviert ist. Wenn ich länger warte, kommt kein Signal vom Pi
The PIR sensor works if I move within 10 seconds if the Pi's HDMI output is disabled. If I wait longer, no signal comes from the Pi

ThSGM, 2021-09-19:
> I had the same problem with my PIR behind the mirror film/glass. I rectified by using a different PIR above the mirror (as the previous was fixed in place), perhaps either a faulty PIR, or that the [...]

I'm encountering, what I think, is a very similar problem: when the HDMI Off is signaled, my LCD screen turns off then immediately back on. I moved my RCWL-0516 doppler sensor to further away from the LCD and other components and this behaviour stopped. 

Just to be clear, I'm not sure this is an issue with the sensor itself, since when I turn `powerSaving` to off the sensor indicators work perfectly---they trigger on when movement is detected, and trigger off after a slight delay. 

```
{
                        module: 'MMM-PIR-Sensor',
                        position: "bottom_right",
                        config: {
                                sensorPin: 22,
                                powerSaving: false,
                                powerSavingDelay: 10,
                                presenceIndicator: "fa-eye",
                                presenceOffIndicator: "fa-eye", // Customizing the indicator
                                presenceIndicatorColor: "#f51d16", // Customizing the indicator
                                presenceOffIndicatorColor: "#3f78a1", // Customizing the indicator
                        }
                },
```

I think this is an issue where the sensor is picking up movement too quickly after the LCD screen is turned off, maybe so that the act of switching off the LCD screen causes the sensor to trigger. 

@rpinyxxjoe, @Skorpionbird, @Corburn84, @renaudjc  : can you all clarify what type of sensor you're using? 

Does anyone spot an easy way to put in a delay after the monitor is turned off? It could be all that is necessary is to make a new option `powerOnDelay` which prevents the monitor from switching on after it has switched off?