paviro/MMM-PIR-Sensor #28: epoll compiled for the wrong version of Node?

jwu910, 2017-04-12:
Hi, i've tried reinstallng and rebuilding epoll, with no success. Has anyone else run in to this issue? Seems like the only problem i'm running in to.

Thanks!

Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build/Release/epoll.node'
0|mm       | was compiled against a different Node.js version using
0|mm       | NODE_MODULE_VERSION 48. This version of Node.js requires
0|mm       | NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
0|mm       | the module (for instance, using `npm rebuild` or`npm install`).
0|mm       |     at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
0|mm       |     at Object.Module._extensions..node (module.js:598:18)
0|mm       |     at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
0|mm       |     at Module.load (module.js:488:32)
0|mm       |     at tryModuleLoad (module.js:447:12)
0|mm       |     at Function.Module._load (module.js:439:3)
0|mm       |     at Module.require (module.js:498:17)
0|mm       |     at require (internal/module.js:20:19)
0|mm       |     at bindings (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bindings/bindings.js:76:44)
0|mm       |     at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/epoll.js:1:192)

jwu910, 2017-04-13:
I may have found a solution. Looked up options to try and use an older version of node and came across instructions to run 

> sudo npm install --save-dev electron-rebuild
> ./node_modules/.bin/electron-rebuild

This got rid of the error message and now i'm working through getting this up.

Wippo0, 2017-04-17:
Same issue, but didn't get rid of the error by the description above

haelbichalex, 2017-04-25:
same issue for me. @jwu910 solution didn't work for me either.

```
WARNING! Could not load config file. Starting with default configuration. Error found: Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build/Release/epoll.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 51. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
```

jwu910, 2017-04-28:
@haelbichalex @Wippo0 I looked a lot in to epoll and tried uninstalling and reinstalling it.

I think i tired npm uninstall epoll, as well as npm uninstall electron and went forward again with the rebuild with the --save tag

rjhornsby, 2017-06-16:
Any additional hints? Have re-installed my RPi2B from scratch and followed the most basic instructions for getting MM with MMM-Button working - and still ran into the issue described here again.

balassy, 2019-01-04:
I've run into the same issue today with MagicMirror `2.7.0-develop`, and I could successfully resolve it with these steps:
```bash
cd modules/MMM-PIR-Sensor
rm package-lock.json
rm -r ./node_modules
npm i
```
Hope this helps!

Manu206, 2019-09-18:
For me none of the solutions worked. However I solved it now

1. Navigate to the MMM-PIR-Sensor folder
2. Run `npm install --save-dev electron-rebuild`
3. Remove the node-modules folder, as well as the packages-lock.json file.
4. Run npm install
5. Run ./node_modules/.bin/electron-rebuild to rebuild everything

Now it is working. Hope this helps ;)

balassy, 2019-09-19:
@Manu206 Thanks for sharing your solution!