paviro/MMM-PIR-Sensor #29: Nothing Happening - Monitor not turning off

Rhuen, 2017-04-27:
Hello all, I am new to programming and this is my first project. I have been unable to get this module working. I went through the installation steps many times over again to try and fix the issue. I am not getting errors. My sensor is working fine as I tested it earlier. Whenever I start up Magic Mirror nothing happens in regards to this module. It's just the regular Magic Mirror screen and the LED I am using show that the sensor detects movement when I wave my hand but no change in the display.

fry0815, 2017-04-28:
Hi Rhuen, can you please open a Terminal at the Pi an execute this command: /opt/vc/bin/tvservice -s 
and post the output here. Maybe with sudo, i don't know.

maxtone66, 2017-05-01:
Hi,
I've got the same problem as Rhuen. I'm sur the PIR sensor works by testing a python script.
Also when I test your module, nothing happen. I followed all the instructions.
For me the command give me this :

pi@mirror:~ $ /opt/vc/bin/tvservice -s
state 0x12000a [HDMI DMT (82) RGB full 16:9], 1920x1080 @ 60.00Hz, progressive

Thanks for your help

fry0815, 2017-05-03:
Hi maxtone66,
your state ist correct, you can see the stats here: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=52309
Can you check youuse the correct Pinnumber? You should use the BCM-numbering: https://de.pinout.xyz/
I use pin 12 on my raspberry, so i should use 18 in the config.

hans2012, 2017-06-12:
hi, i got the same problem, can somebody pls check my module insertion...
modules: [
                {
                        module: "alert",
                },
                {
                        module: "MMM-PIR-Sensor",
                        config: {
                        sensorPIN: "25",
                        powerSaving: "true"
                                }
                },

cyberdie, 2017-06-12:
Same problem here with pin 15 (22 in conf.js)

fry0815, 2017-06-13:
Hi @hans2012,
try it without the "
For example:

module: "MMM-PIR-Sensor",
config: {
    sensorPIN: 25,
    powerSaving: true
}

cyberdie, 2017-06-13:
In my case the config is without quotes:

`{
                        module: 'MMM-PIR-Sensor',
                        config: {
                                sensorPIN: 22,
                                powerSaving: true
                                }
                },
`

hans2012, 2017-06-14:
tnx , ill try it when i get home, ill post the result.

hans2012, 2017-06-14:
still not working, only the led from the sensor will turn red, but the monitor stays on...i read other modules sometimes they used " ", others used this ' '  i tried both but still not working for me. the any other suggestions...

Rutul27, 2018-12-08:
I hope this is not very late for the solution, but there is a small mistake in the config. Please try sensorPin instead of sensorPIN or copy paste the following:

{
module: "MMM-PIR-Sensor",
config: {
sensorPin: "22",
powerSaving: "true"
}
},

@Rhuen @fry0815 @maxtone66 @hans2012 @cyberdie