paviro/MMM-PIR-Sensor #32: Monior wakes up But screen is black

Zimmerjo, 2017-06-27:
I've updated my Magic Mirror to the newest version.
Bevor eveything was great and worked.
Now I got the problem that after sleeping the Monitor turns on but the screen is black.
I changed nothing more. Dou you have some ideas, whats wrong?
Thank you 

If i check with `tvservice -s`
I got these answer when it's on:
`state 0x120006 [DVI DMT (83) RGB full 16:9], 1600x900 @ 60.00Hz, progressive`
and if it's off these:
`state 0x120002 [TV is off]`

bwente, 2017-08-11:
Brand new install last night and I also get a black screen. Comment out the the config and it works. I can also issue the tvservice commands and the monitor responds accordingly.

Zimmerjo, 2017-09-09:
Thx for your advice. That's the way a I did but the Mointor is now always on... Dou you have another way to fix that ?

godinperson, 2017-10-06:
Same issue here! After the module installation, my monitor is black. If I VNC on it, I can see my mouse

SonnyX87, 2017-10-17:
Hey, I had the same issue.
Please replace the commands with `exec("sudo vcgencmd display_power 1", null);` to turn on
and `exec("sudo vcgencmd display_power 0", null);` to turn it off.

rowe2, 2017-12-04:
Hi, I got the same problem on a raspy zero.

Where do I have to raplace this commands?

SonnyX87, 2017-12-05:
Hi,
replace it like below.

File: node_helper.js 
Line 25 to 28
`exec("sudo vcgencmd display_power 1", null);`

File: node_helper.js 
Line 37
`exec("sudo vcgencmd display_power 0", null);`

jinumohan173, 2018-02-15:
new version of magic mirror having issue with this...older version was perfectly working

ramymounir, 2020-11-06:
> Hi,
> replace it like below.
> 
> File: node_helper.js
> Line 25 to 28
> `exec("sudo vcgencmd display_power 1", null);`
> 
> File: node_helper.js
> Line 37
> `exec("sudo vcgencmd display_power 0", null);`

ramymounir, 2020-11-06:
how do I edit the file node_helper.js? to replace the linens you are mentioning?

ramymounir, 2020-11-07:
I really cant understand how can someone write such an unclear message lol.

please replace the commands? why commands, where and how? how do I open the node helper.js and replace the commands?
can you give us some steps?

SonnyX87, 2020-11-09:
> I really cant understand how can someone write such an unclear message lol.
> 
> please replace the commands? what commands, where and how? how do I open the node helper.js and replace the commands?
> can you give us some steps?

Hey, my comment is almost three years old!! I don't know if it still works. 

And what kind of steps? How to find and open the file? The mentioned file is in the root of this project. You can open it for example with Visual Studio Code or another editor. Whether the lines I have mentioned are still correct, I have not checked now. But where you have installed MagicMirror, I don't know. Anyway, the extensions belong in the folder modules, which is located in the installation folder.

As I said, my solution is almost three years old. Whether the lines I have mentioned are still correct, I have not checked now. As I said, my solution is almost three years old and might not work at all.

ramymounir, 2020-11-09:
It doesn’t mate 
I tried 
Unfortunately not

ArtofWar-dev, 2021-07-02:
Is there any fix to this yet? I'm having the same issue.