paviro/MMM-PIR-Sensor #33: Problem with MM update 2.1.2

Fonfonof, 2017-07-03:
Hi,

After the last updating of MM in 2.1.2 i have an issue like this:

WARNING! Could not load config file. Starting with default configuration. Error found: Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build/Release/epoll.node’
was compiled against a different Node.js version using
NODE_MODULE_VERSION 50. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using npm rebuild ornpm install).
0

how i can fix it ?

thanks

PtrBld, 2017-07-12:
Hi try this like stated in #9 

Execute the following commands in /pi/magicmirror/ (your magicmirror installation folder):

npm uninstall electron
npm rebuild
npm install electron

semox, 2017-08-10:
This does not work on my side. Is there any workaround?

semox, 2017-08-11:
This did the trick for me:
`cd ~/MagicMirror
git pull
rm -rf node_modules/
npm cache clean
npm install
for f in `find /home/pi/MagicMirror/modules -maxdepth 1 -name "MMM-*" -type d`; do cd $f; git pull && npm install; done`

metalshreds, 2017-08-24:
I tried both suggestions from PtrBld and semox twice to see if it would fix it and unfortunately It didn't, is there anything else to try? running latest version of magic mirror

konecnyna, 2017-09-09:
Also having the same problem.

CaptainFlam27, 2017-09-10:
Hi everybody !

Also having the same problem. I'm a newbie regarding rasperry pi & magic mirror modules. I did the magic mirror setup some time ago and everything worked well. I let it aside for some time and reopened it only this week-end. I did the update as suggested by the notification on the mirror and now got this error when launching : 

App threw an error during load
Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/e                     poll/build/Release/epoll.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 53. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:                     20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20                     )
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at bindings (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bin                     dings/bindings.js:76:44)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_m                     odules/epoll/epoll.js:1:192)
Whoops! There was an uncaught exception...
Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/e                     poll/build/Release/epoll.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 53. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:                     20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20                     )
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at bindings (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bin                     dings/bindings.js:76:44)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_m                     odules/epoll/epoll.js:1:192)

I tried the PtrBld fix but with no effects, I haven't tried semox fix yet

cyberdie, 2017-09-11:
Same here.
Tried any proposed solution here with no effects.
I think this module is broken for current version of MM.

semox, 2017-09-11:
Hi,

could someone try:

'cd /home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules
npm rebuild'

metalshreds, 2017-09-11:
@semox that worked, I tried that and get no errors when running magic mirror. My problem now is that my sensor doesn't appear to be working. Thanks!

metalshreds, 2017-09-12:
I was able to get my sensor working, but when the screen wakes up for the first time all I see is a blinking cursor on the top left of the screen. I don't get any errors on the terminal, and if I unplug the PIR sensor magic mirror works as expected, I am using pin 4 BCM. Maybe somebody else can see if this is happening to them as well.

CaptainFlam27, 2017-09-24:
Hi everybody, 
I tried @semox new solution : 

> 'cd /home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules
> npm rebuild'

but unfortunately I still got the same problem :

> WARNING! Could not load config file. Starting with default configuration. Error found: Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build/Release/epoll.node'
> was compiled against a different Node.js version using
> NODE_MODULE_VERSION 48. This version of Node.js requires
> NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
> the module (for instance, using `npm rebuild` or`npm install`).

simpleex, 2017-10-09:
Same problem here (I tried all the fixes above) !

```
WARNING! Could not load config file. Starting with default configuration. Error found: Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build/Release/epoll.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 48. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using npm rebuild ornpm install).
```

metalshreds, 2017-10-12:
seems like it was a temporary fix, and never got the PIR sensor to work, I booted my magic mirror today and got the same error. commented out the section in the config file and the magic mirror booted fine. 

`WARNING! Could not load config file. Starting with default configuration. Error found: Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build/Release/epoll.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 54. This version of Node.js requires
NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).`