paviro/MMM-PIR-Sensor #39: MM v2.2.1 error at starting

Snake16547, 2018-01-01:
Heyho,
when running the MMM-PIR-Sensor Module on the new 2.2.1 MM Version i got following error message at **npm start dev**

```
Initializing new module helper ...
Module helper loaded: MMM-temp-ds18b20
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: newsfeed
App threw an error during load
Error: Module version mismatch. Expected 50, got 54.
    at Error (native)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:583:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
    at bindings (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bindings/bindings.js:76:44)
Whoops! There was an uncaught exception...
Error: Module version mismatch. Expected 50, got 54.
    at Error (native)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:583:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
    at bindings (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bindings/bindings.js:76:44)

```

qistoph, 2018-01-03:
Hi @Snake16547 ,

This is an issue that has been popping up once in a while with the module. A quick fix for your situation:

```bash
#Change into the modules directory
cd modules/MMM-PIR
# Rebuild module for the correct elecron version
npm rebuild --runtime=electron --target=$(../../node_modules/electron/cli.js --version) --disturl=https://atom.io/download/atom-shell --build-from-source
```

@paviro and any other contributors, since this issue keeps coming back I would appreciate a more reliant and future proof solution...

mnefk, 2018-01-03:
thanks qistoph but this give's me the follow errors
`> epoll@0.1.22 install /home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll
> node-gyp rebuild

gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: 404 response downloading https://atom.io/download/atom-shell/v6.12.3/node-v6.12.3-headers.tar.gz
gyp ERR! stack     at Request.<anonymous> (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:210:14)
gyp ERR! stack     at emitOne (events.js:101:20)
gyp ERR! stack     at Request.emit (events.js:188:7)
gyp ERR! stack     at Request.onRequestResponse (/usr/lib/node_modules/npm/node_modules/request/request.js:954:10)
gyp ERR! stack     at emitOne (events.js:96:13)
gyp ERR! stack     at ClientRequest.emit (events.js:188:7)
gyp ERR! stack     at HTTPParser.parserOnIncomingClient (_http_client.js:474:21)
gyp ERR! stack     at HTTPParser.parserOnHeadersComplete (_http_common.js:99:23)
gyp ERR! stack     at TLSSocket.socketOnData (_http_client.js:363:20)
gyp ERR! stack     at emitOne (events.js:96:13)
gyp ERR! System Linux 4.9.35-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll
gyp ERR! node -v v6.12.3
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok

npm ERR! Linux 4.9.35-v7+
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "rebuild" "--runtime=electron" "--target=" "--disturl=https://atom.io/download/atom-shell" "--build-from-source"
npm ERR! node v6.12.3
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! epoll@0.1.22 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the epoll@0.1.22 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the epoll package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs epoll
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls epoll
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/modules/MMM-PIR-Sensor/npm-debug.log`

Snake16547, 2018-01-03:
in my case this helped

`npm rebuild --runtime=electron --target=1.4.5 --disturl=https://atom.io/download/atom-shell --abi=50
`

Milkdog, 2018-01-04:
I just had this issue. To resolve it, I deleted the `node_modules` folder from within the module, and just re-ran `npm install`