paviro/MMM-PIR-Sensor #40: powerSavingDelay not working

skaborg, 2018-03-10:
powerSavingDelay not working!
my config:
{
 module: 'MMM-PIR-Sensor',
  config: {
   powerSavingDelay: 30,
   sensorPIN: "3",
  },
},
After restart MMM, monitor is blank (no signal) - it's OK.
if i'm closed pin GPIO 3 (pin 15 in rpi3) - monitor turn on - and I can see MMM screen. It's OK.
After i'm open pin GPIO 3, about 9-10 sec monitor turn off! always!!! 
regardless of the parameter powerSavingDelay settings!

skaborg, 2018-03-10:
after reboot module MMM-PIR-Sensor not working!
monitor all time is on.
I type in console (pin GPIO3 is open)
gpio read 3
0
If pin GPIO3 is closed
gpio read 3
1

tvservice -s
state 0x120006 [DVI DMT (35) RGB full 5:4], 1280x1024 @ 60.00Hz, progressive

tvservice -o
Powering off HDMI
(monitor turn off)

tvservice -s
state 0x120002 [TV is off]

 tvservice --preferred && chvt 6 && chvt 7
Powering on HDMI with preferred settings
(monitor turn on)
but control on|off with module MMM-PIR-Sensor not working

flyingchipmunk, 2018-06-20:
Pull request #48 might be related fix. my tvservice inside this module also did not work, switched it out with vcgencmd calls.

skaborg, 2018-12-05:
No! It's dont work to!
my config:
              {
                      module: 'MMM-PIR-Sensor',
                      config: {
                              // See 'Configuration options' for more information.
                              sensorPin: 17,
                              relayState: 0,
                              powerSaving: true,
                              powerSavingDelay: 60,
                              powerSavingNotification: false
                      }
              },
if i'm open pin BCM 17 (pin 11 in rpi3) - monitor turn off after 60 sec. It's OK.
If monitor is off and I'm close pin BCM 17 - monitor turn on. It's OK.
If monitor is on and I'm open pin BCM 17 and close it again after 30 sec (for example) - the timer is not reset and monitor turn off after 30 sec all the same (60 seconds after first clack)!!!  It's Bug!!!