paviro/MMM-PIR-Sensor #44: MagicMirror doesn't start with this module

delema98, 2018-05-03:
Hi, 
i'm totally new to the MagicMirror. When I added this module, there's always this error: 

WARNING! Could not load config file. Starting with default configuration. Error                                                                                                                                                              found: Error: Cannot find module 'onoff'.

I installed wiring-pi in the MagicMirror Folder with "npm install wiring-pi".

Has anybody an idea?

thanks

tung256, 2018-05-18:
i just installed MagicMirror from this guide: https://howchoo.com/g/ntcymzbimjv/how-to-install-magic-mirror-on-your-raspberry-pi

everything is working BUT this pir motion module. 
if i remove the module from MagicMirror/config/config.js,  i am back to normal again. if i add the MMM-PIR-Sensor, i get mouse and black screen.  someone pls help us.

flyingchipmunk, 2018-06-20:
What does the output of "pm2 log MagicMirror" look like?

Jouhney, 2020-08-04:
I had the same problem. I navigated to the MMM-PIR-Sensor module folder ( cd ./MagicMirror/modules/MMM-PIR-Sensor) and executed "npm install". After that, the MagicMirror started working again