paviro/MMM-PIR-Sensor #49: error since updating magic mirror to version 01.07.2018

apm1007, 2018-07-01:
get the following error message when npm install:
pi@raspberrypi:~/MagicMirror/modules/MMM-PIR-Sensor $ npm install

> Magic-Mirror-Module-PIR-Sensor@1.1.0 postinstall /home/pi/MagicMirror/modules/MMM-PIR-Sensor
> electron-rebuild -e ../../node_modules/electron

⠋ Searching dependency tree
An unhandled error occurred inside electron-rebuild
Could not detect abi for version 2.0.3 and runtime electron.  Updating "node-abi" might help solve this issue if it is a new release of electron

Error: Could not detect abi for version 2.0.3 and runtime electron.  Updating "node-abi" might help solve this issue if it is a new release of electron
    at Object.getAbi (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/node-abi/index.js:30:9)
    at new Rebuilder (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/rebuild.js:64:28)
    at rebuildWithOptions (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/rebuild.js:310:23)
    at Object.doRebuild [as rebuild] (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/rebuild.js:317:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/cli.js:114:33)
    at next (native)
    at fulfilled (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/cli.js:5:58)

npm ERR! Linux 4.14.50-v7+
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! node v6.14.3
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! Magic-Mirror-Module-PIR-Sensor@1.1.0 postinstall: `electron-rebuild -e ../../node_modules/electron`
npm ERR! Exit status 255
npm ERR!
npm ERR! Failed at the Magic-Mirror-Module-PIR-Sensor@1.1.0 postinstall script 'electron-rebuild -e ../../node_modules/electron'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the Magic-Mirror-Module-PIR-Sensor package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron-rebuild -e ../../node_modules/electron
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs Magic-Mirror-Module-PIR-Sensor
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls Magic-Mirror-Module-PIR-Sensor
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/modules/MMM-PIR-Sensor/npm-debug.log
[npm-debug.log](https://github.com/paviro/MMM-PIR-Sensor/files/2153205/npm-debug.log)
 
Starting Magic Mirror
App threw an error during load
Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build/Release/epoll.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 48. This version of Node.js requires
NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:172:20)
    at Object.Module._extensions..node (module.js:671:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:172:20)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (module.js:586:17)
    at require (internal/module.js:11:18)
    at bindings (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bindings/bindings.js:81:44)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/epoll.js:1:192)
Whoops! There was an uncaught exception...
Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build/Release/epoll.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 48. This version of Node.js requires
NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:172:20)
    at Object.Module._extensions..node (module.js:671:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:172:20)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (module.js:586:17)
    at require (internal/module.js:11:18)
    at bindings (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bindings/bindings.js:81:44)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/epoll.js:1:192)
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
Launching application.

Thanks for your help.

apm1007, 2018-07-02:
> I fixt it,
>
> Go to the MagicMirror/modules/MMM-PIR-Sensor/ dir
> and do npm install

No. I did Install ans rebuild. Thema Error still exists

Duske, 2018-07-04:
I needed to delete the `node_modules` directory first. Then I ran `npm install` and the issue was gone.

apm1007, 2018-07-04:
Thank You very much. This fixed IT.

lodesmets, 2018-08-04:
If this fixed your issue, please mark it as closed

closed 2018-08-05 by apm1007