paviro/MMM-PIR-Sensor #54: blank screen after adding the module

mghiraw, 2018-07-20:
Hi
After adding the module Iam getting a blank screen.
See the error log below.

/home/pi/.pm2/logs/mm-error.log last 15 lines:
0|mm       |     at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_helper.js:148:3)
0|mm       |     at Module._compile (module.js:571:32)
0|mm       |     at Object.Module._extensions..js (module.js:580:10)
0|mm       |     at Module.load (module.js:488:32)
0|mm       | { Error: Cannot find module 'onoff'
0|mm       |     at Module._resolveFilename (module.js:470:15)
0|mm       |     at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
0|mm       |     at Function.Module._load (module.js:418:25)
0|mm       |     at Module.require (module.js:498:17)
0|mm       |     at require (internal/module.js:20:19)
0|mm       |     at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_helper.js:11:14)
0|mm       |     at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_helper.js:148:3)
0|mm       |     at Module._compile (module.js:571:32)
0|mm       |     at Object.Module._extensions..js (module.js:580:10)
0|mm       |     at Module.load (module.js:488:32) code: 'MODULE_NOT_FOUND' }

lodesmets, 2018-08-05:
I had the same problem.
I fixed it by navigating in the MMM-PIR-Sensor folder and then npm install

xplod609, 2020-05-10:
this helped me 
thanks