paviro/MMM-PIR-Sensor #65: Must have module - but complicated

shinyone, 2018-09-28:
There is a lot of interest from Magic Mirror builders in this module, since it's such a cool and easy idea - in principle.

However, the forums and websites are filled with questions on how to get this working.   I find myself in the same situation and have being overcoming problems with this module.   I'm tempted to write and troubleshooting guide for the main README.

1)  Rasp Pi 3b+  -  First issue.   [WiringPi](http://wiringpi.com/download-and-install/).   Requires version is v2.46 to avoid `Unable to determine hardware version` using command `gpio -v` Requires:  latest version
2)  [EPOLL ](https://github.com/paviro/MMM-PIR-Sensor/issues/28) - Problem with epoll complied for wrong version of node "was compiled against a different Node.js version using". 
Requires command:  `sudo npm install --save-dev electron-rebuild ./node_modules/.bin/electron-rebuild`
3) ELECTRON - Faced a black screen after last step.  Requires recompile of new version.  [Good Thread](https://forum.magicmirror.builders/topic/1351/mmm-pir-sensor-install-failing/62)

> npm list | grep electron
> npm rebuild --runtime=electron --target=2.0.0--disturl=https://atom.io/download/atom-shell --abi=50 
> 
4)  Looked at pm2 logs, no problems reported  `pm2 logs /mm/`

I have used various python scripts to check that the sensor is working.  It is.

I'm currently using the follow python script as a service.  Monitor turns on and off with motion
https://github.com/deg0nz/MagicMirror-Display-Controller

However, I really want this module to work so that I can:

- listen to the USER_PRESENCE notification in my own module

So finally the question, how can I debug further?  

- How can I go from a working python script to a working module?

- What config settings can be used to force the module to show results quickly?

shinyone, 2018-09-28:
Quick update .

After adding a console.log  in [node_helper.js](https://github.com/paviro/MMM-PIR-Sensor/blob/master/node_helper.js#L112)

> this.pir = new Gpio(this.config.sensorPin, 'in', 'both');

I found that the sensorPin is 22, rather than the value 4, that I placed in config.js.

I then modified the [main ](https://github.com/paviro/MMM-PIR-Sensor/blob/master/MMM-PIR-Sensor.js#L13) file to change the default sensorPin to 4.

All working.

From what I can see the CONFIG notification is sent to the node_helper.js in the start() method of the module.  Not sure where the problem is, but the powerSavingDelay value that I changed in the config.js is correctly passed

shinyone, 2018-09-28:
UPDATE:   Found the problem.   I think I copied the config from another forum post which was incorrect.

~~sensorPIN~~ sensorPin: 4,

``` JS
                module: 'MMM-PIR-Sensor',
                config: {
                        sensorPin: 4,
                        powerSaving: true,
                        powerSavingDelay: 60
                }

```

shinyone, 2018-09-28:
Leaving open for reference

metalshreds, 2018-09-30:
Thanks for the detailed steps on getting this module working! unfortunately I can't get past the error 
```
Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build/Release/epoll.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
```
I've rebuilt it multiple times with electron-rebuild, deleted the node_modules folder and npm I again, and nothing seems to be working.

Any idea how I could fix it? I'm running it on a raspberry pi w with stretch lite. Thanks in advance for any help :)

https://electronjs.org/docs/tutorial/using-native-node-modules

metalshreds, 2018-10-01:
ok finally was able to some the "compiled against a different Node.js version" by manually installing the epoll module inside from inside the MMM-PIR-Sensor module folder. Nothing else seemed to work, even rebuilding electron 

$ npm install epoll

https://www.npmjs.com/package/epoll

shinyone, 2018-10-03:
thanks for adding the extra information that solved your problem.

CriticalPoint, 2019-01-23:
> ~~ **I think I copied the config from another forum post which was incorrect** ~~
> 

Hmm, I'm not buying into that. For your sanity, I think a version changed caused the capitalisation issue. I've spent (like yourself) a good day or so troubleshooting this, only to find the same issue (was `sensorPIN`, should be `sensorPin`).

Long story short, I just looked back through commits and found this little gem, so don't be too hard on yourself!

[image]

shinyone, 2020-05-01:
Update since I face the a similar problem after update.

Update version to match electron version (Found in the package.json of the main MagicMirror/node_modules/electron directory)

sudo npm rebuild --runtime=electron --target=6.1.7 --disturl=https://atom.io/download/atom-shell --build-from-source