paviro/MMM-PIR-Sensor #69: No HDMI output after 5min

Skorpionbird, 2018-12-25:
I have the following problem. The sensor and shutdown work. When the picture goes out and I move, it turns on again. If no one stands in front of the MaggicMirror for 5 minutes, the HDMI output can no longer be activated and the monitor remains dark. Can someone help me with that?

henrysun18, 2019-01-03:
I'm getting this issue as well. It was working fine a few months ago, which was when I last used the mirror before unplugging my rpi3. Today I noticed that after 5 minutes from when MagicMirror starts up the screen just blanks out and won't display anything. The LCD backlight is still on, just with nothing displayed.

CriticalPoint, 2019-01-23:
I had a _similar_ issue with sleep, I found this - [https://raspberrypi.stackexchange.com/questions/75893/disabling-sleep-on-rasberry-pi3](https://raspberrypi.stackexchange.com/questions/75893/disabling-sleep-on-rasberry-pi3)

You can disable sleep by opening up `sudo nano /etc/lightdm/lightdm.conf`, finding this line `#xserver-command=X` and appending `-s 0 dpms` to the end, so it reads `xserver-command=X -s 0 dpms`.

Reboot - `sudo reboot`

Let us know if this helps in your specific case
/ Crit

DiscGolf, 2020-01-02:
Hi, editing lightdm.conf didn't help. 
xserver-command=X -s 0 dpms
Do I have to uncomment further entries to take affect?