paviro/MMM-PIR-Sensor #74: Setting Up The PIR

dxfan227, 2019-04-03:
Sorry If this is the wrong place to ask. I want to use the MMM-PIR-Sensor Module to turn on my magic mirror....however I am very very new to all of this and am unsure of how to actually connect the PIR to the Pi's GPIO ports.....do you solder? Use some kind of female to female cable?  is there any videos out there or good walk throughs that could help me out>

Jouhney, 2020-07-28:
I have the same question. I found a so-called breadboard, which however is pretty intimidating. There's another option. It's a jumper cable wich, I think, can be plugged in the GPIO ports on the Raspberry Pi.

eric-h-st, 2020-07-28:
A female-female cable, such as this [on Amazon Canada](https://www.amazon.ca/Elegoo-120pcs-Multicolored-Breadboard-arduino/dp/B01EV70C78) should work very well.

Jouhney, 2020-07-28:
Thanks a lot! 👍

closed 2020-07-28 by eric-h-st