paviro/MMM-PIR-Sensor #75: Node.js version for MagicMirror v2.7.1 Error

meiringjohan, 2019-04-03:
Hi,

After upgrading to v2.7.1 of MagicMirror the following error occurs when starting MagicMirror

here is the error I get:

0|mm  | App threw an error during load
0|mm  | Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build/Release/epoll.node'
0|mm  | was compiled against a different Node.js version using
0|mm  | NODE_MODULE_VERSION 59. This version of Node.js requires
0|mm  | NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
0|mm  | the module (for instance, using `npm rebuild` or `npm install`).
0|mm  |     at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
0|mm  |     at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
0|mm  |     at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
0|mm  |     at Module.load (internal/modules/cjs/loader.js:620:32)
0|mm  |     at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
0|mm  |     at Function.Module._load (internal/modules/cjs/loader.js:551:3)
0|mm  |     at Module.require (internal/modules/cjs/loader.js:658:17)
0|mm  |     at require (internal/modules/cjs/helpers.js:20:18)
0|mm  |     at bindings (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bindings/bindings.js:81:44)
0|mm  |     at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/epoll.js:1:192)

meiringjohan, 2019-04-03:
Found a solution!

in terminal browse to: ~/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll

Then run the following comand:  npm rebuild

This will rebuild epoll.node with the correct version.

SdeGeata, 2019-06-27:
That worked like a charm! Thanks meiringjohan!!!