paviro/MMM-PIR-Sensor #76: MMM-PIR-Sensor not working with latest release - display does not turn off

Kugelfang666, 2019-04-04:
Dear all,

I’ve been using MM for over 1,5 years now. Unfortunately my SD card died yesterday so I took the opportunity to start from scratch.

I installed the latest release 2.7.1 on my RBPi 3. using the “Full KMS” OpenGl driver. Unfortunately my PIR Sensor does not work anymore as my display stays on all the time. Interestingly I’m able to turn the display on an off manually via SSH using

`cgencmd display_power 0`
`vcgencmd display_power 1`

I already checked the wiring (which I actually did not change from my setup which was working for a long time) and I also checked the signal line of the PIR Sensor. Both are ok.

My configuration looks like this:

```
		{
			module: 'MMM-PIR-Sensor',
			config: {
				sensorPIN: 23,
				powerSavingDelay: 15
			}
		},
```
It would be great if someone could help me out as I’m clearly missing something…

Kugelfang666, 2019-04-04:
actually I cheeped the pm2 logs and found loots of this messages:

```
npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-04-03T22_01_36_315Z-debug.log
/home/pi/MagicMirror/node_modules/electron/dist/electron js/electron.js: symbol lookup error: /home/pi/MagicMirror/modules/MMM-PIR/node_modules/epoll/build/Release/epoll.node: undefined symbol: _ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEiNS_19ConstructorBehaviorENS_14SideEffectTypeE
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! magicmirror@2.7.1 start: `sh run-start.sh`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the magicmirror@2.7.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

FatTony123, 2019-04-14:
I'm having a similar issue. Updated the MMM-PIR-Sensor repo, magicmirror software and my raspberry pi, somehow the module now keeps my display dark once started. 

Checked my HDMI signal with the tvservice command and it's showing power on it correctly. 
Also my PIR is receiving input when tested with a separate script. 

I commented out the MMM-PIR-Sensor module for now. 
Once using pm2 restart mm, the screen starts correctly.

Kugelfang666, 2019-04-14:
had the same problem. Finally switched to MMM-PIR instead of this module

FatTony123, 2019-04-15:
Ok, will give it a try. Need to switch on / off the display using a relay, not sure MMM-PIR can do that

pixxelfriend, 2020-07-24:
> I'm having a similar issue. Updated the MMM-PIR-Sensor repo, magicmirror software and my raspberry pi, somehow the module now keeps my display dark once started.
>
> Checked my HDMI signal with the [...]

I had the same problem after MM update. When MM-PIR-Sensor was activated, i just saw a black screen. I solved it by, removing the `node_modules` folder from the module itself and running a fresh npm install.
`
~/MagicMirror/modules/MMM-PIR-Sensor $ rm -R node_modules/
~/MagicMirror/modules/MMM-PIR-Sensor $ npm install
`