paviro/MMM-PIR-Sensor #77: Magic Mirror restart auto

tiruds76, 2019-04-08:
hello i have a problem with the sensor everything works okay but as soon as it has cycled it off the screen and turns on itself after 5 to 10 seconds do you have any idea of the problem?

Jann86, 2019-05-24:
I have the same Issue....

tanvir586, 2019-10-14:
I have same aswell.  Anyone can help?

samsnv, 2019-12-10:
yes the same prblm !!!

tanvir586, 2019-12-10:
Ive figure out the problem.  It's the sensor problem not a module problem.
First i used this sensor 
[image]

and it turns off the screen and turns on itself after 5 to 10 seconds. I thought maybe my sensor is faulty. So i bought same sensor again but no luck. 
Then i bought this one....
Oiyagai 2pcs AM312 Mini Pyroelectric PIR Sensor Module Manual Motion Infrared IR Detector https://www.amazon.co.uk/dp/B07FGG87JM/ref=cm_sw_r_cp_apa_i_Q7b8DbNZ75MBS
And worked pretty decent for me.