paviro/MMM-PIR-Sensor #79: When the pi receives no signal for extended time....it closes all apps and basically becomes unresponsive

dxfan227, 2019-04-21:
I am attempting to turn on/off my display via a motion detector. The module fires the commands vcgencmd display_power 0/1. this works fine except that once the monitor is off with no signal for a while.....the pi seems to go into some type of sleep mode or something.

The pi is still on but all apps are closed and it will not turn on via motion or keyboard info....it will turn on by manually sending the vcgencmd display_power command however... all the apps are closed including my motion detector anyone have any thoughts.

M08IUS0NE, 2019-09-12:
I'm having something similar. Everything works great, responds great, but once I walk away and don't interact for more than 5 or so minutes, when I walk in front to activate the system I get nothing. No response, no screen, no keyboard response. Screen stays blank and the only thing I can do is pull the plug and restart everything. I've tried everything that I've seen suggested:
- installed xscreensaver
- added the three command lines to the LXDE-pi/autostart file (were already on there when I got to it, possibly by xscreensaver)
- modified the lightdm.conf file

So far no luck. Not sure if this is the same issue that you're experiencing but I've tried all recommendations and have had no luck. I am currently running Buster, so I wonder if maybe the newer version is to blame? Any help would be appreciated.

dxfan227, 2019-09-12:
Hey im sorry i dont have a fix for you. I did do something to make it work
better but it's been so long I cant remember :(. I think i went with either
a different addon maybe ? :( I try and boot the machine up and see if I can
remember. I havent had a chance to finish my project yet.

Thanks,

David B. Ruiz |
*Director of Recruitment**G*
*lobal Talent Solutions, Inc.**Email:* druiz@gtsrecruiters.com
*Office:* 714-921-6428 x. 307
*Mobile:* (408) 641-9701
*Web:* http://www.GTSrecruiters.com <http://www.gtsrecruiters.com/>