paviro/MMM-PIR-Sensor #80: how to make other modules wake monitor up

milkywade, 2019-06-01:
Hi - I have another module running in the MM that generates sound alerts every now and then. The sound is played via the monitor built-in speakers. However, If the monitor is powered down, no sound comes out of it and the alert is missed.

How can I wake the monitor up following an alert generated by another module ?

thanks!

milkywade, 2019-07-28:
Is this module maintained ?

milkywade, 2019-08-12:
hallo??

BLoodConstructions, 2019-08-12:
For my next MM i‘m planning to use a PIR-Sensor for Display on/off. I would prefer to use a simple python oder shell script of motion detected display on and Play soundfile. I‘ve been watching this module since a year and always read this doesn‘t work etc... So I decided to take an easier way.

milkywade, 2019-08-16:
good idea although it somewhat amounts to rebuilding this module :-)
I ended up adding a separate monitor wake up call to solve my issue as this module maintenance is totally dead.

write2warriors, 2020-05-23:
@milkywade , would you like to share how you accomplished that?