paviro/MMM-PIR-Sensor #90: Electron version number unable to find

haliszebi, 2020-02-18:
Hi!
I'm totally new to the Raspberry Pi (I have the Pi 3 model B+) and I've got a problem with this module. It shows nothing just black screen when it tries to run MagicMirror after I included the config.js. I've seen a comment where the guy said that he npm installed the PIR-Sensor in the directory.. So i've tried it and i've got an ERROR message which I attached. 
[image]

andersarstrand, 2020-04-29:
Similar problem after updating MagicMirror to 2.11.0 .

My solution was :
Go to the module folder (cd modules/MMM-PIR-Sensor )
Reset GIT (git reset --hard) 
Remove the "node_modules" folder 
Reinstall the module (npm install)

txuma2, 2020-05-10:
Same problem but has not been solved when doing the indicated operation
Any ideas?

txuma2, 2020-05-10:
[image]

closed 2020-07-24 by eric-h-st

Coumodo, 2022-01-04:
[image]

seems to be the same/compareable issue - git reset and also complete new MM installation didn't worked.