paviro/MMM-PIR-Sensor #91: Black screen after update

siggmur, 2020-04-02:
Updated MM. 

When I restart MM the screen is black, if I remove MMM-PIR-Sensor from loading everything works fine

rabbit83ka, 2020-04-02:
It's due to an electron-rebuild error, but I don't know how to solve it. Error Message:

> An unhandled error occurred inside electron-rebuild
> Could not detect abi for version 6.1.7 and runtime electron.  Updating "node-abi" might help solve this issue if it is a new release of electron
> 
> Error: Could not detect abi for version 6.1.7 and runtime electron.  Updating "node-abi" might help solve this issue if it is a new release of electron
>     at Object.getAbi (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/node-abi/index.js:30:9)
>     at new Rebuilder (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/rebuild.js:122:28)
>     at rebuildWithOptions (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/rebuild.js:393:23)
>     at Object.doRebuild [as rebuild] (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/rebuild.js:400:16)
>     at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/cli.js:115:33)
>     at Generator.next (<anonymous>)
>     at fulfilled (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/cli.js:5:58)

ChrizZz90, 2020-04-02:
you need to update node-abi. I did that with
`npm install node-abi`
After that go to the MMM-PIR-Sensor folder and use
`npm install`

After that it should work.

Andoramb, 2020-04-08:
I did a fresh install of the module... removed the folder and done the setup again. it worked for me

closed 2020-04-08 by siggmur