paviro/MMM-PIR-Sensor #94: error message

erkilic, 2020-06-10:
hi
after installing your MMM-PIR-Sensor Iget following message after npm start:
----------------------------------------------------------------------------------------------
/home/pi/MagicMirror/node_modules/electron/dist/electron js/electron.js: symbol lookup error: /home/pi/node_modules/epoll/build/Release/epoll.node: undefined symbol: _ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEiNS_19ConstructorBehaviorENS_14SideEffectTypeE
npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! magicmirror@2.10.1 start: `./run-start.sh`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the magicmirror@2.10.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-06-10T08_01_54_038Z-debug.log
----------------------------------------------------------------------------------------------

Any idea, what the reason can be?
I believe it has to do with 
"const Gpio = require('onoff').Gpio;" in node_helper.js

Thanks in advance.
Gürkan

eric-h-st, 2020-07-15:
@erkilic could you share that log file?
Did you run npm install for the MMM-PIR-Sensor?

eric-h-st, 2020-07-24:
Hi.
I just had to solve that for myself after updating Raspbian. I tried all kinds of options until I got to the solution.
Here's how I solved it:
````
cd modules/MMM-PIR-Sensor
npm install --save-dev electron-rebuild
rm package-lock.json
rm -rf ./node_modules
npm install
./node_modules/.bin/electron-rebuild 
````

closed 2020-08-04 by eric-h-st