paviro/MMM-PIR-Sensor #95: PIR Sensor not working

drk-geek, 2020-06-11:
Everything is fine but this module is not working.... tried changing the hardware (PIR Sensor) but still the same.... any fix/how to check status etc?

drk-geek, 2020-06-15:
anybody to respond ?

lxne, 2020-06-15:
I did not get this module to work. Others did.
https://github.com/mboskamp/MMM-PIR was the one that worked for me.
MMM-GoogleAssistant in combination with MMM-Assistant2Display has another PIR module integrated which works fine for me.

drk-geek, 2020-06-17:
hi @lxne do you have any how to? what to do with callbckScripts ?

lxne, 2020-06-17:
> hi @lxne do you have any how to? what to do with callbckScripts ?

Sorry, no. If you consider using GoogleAssistant I absolutely recommend MMM-GoogleAssistant in combination with MMM-Assistant2Display. I didn't have any issues with its integrated PIR module.

In the MMM-PIR is a pretty long [issue how others got it to work](https://github.com/mboskamp/MMM-PIR/issues/10). There is even a raspberry image with a working system you could try.

drk-geek, 2020-06-17:
hello @lxne so you recommend using MMM-GoogleAssistant in combination with MMM-Assistant2Disply, right? 
please suggest recommended how to for both modules as I love to try it...
Thanks in advance

lxne, 2020-06-17:
MMM-GoogleAssistant https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation
MMM-Assistant2Display https://github.com/bugsounet/MMM-Assistant2Display/wiki

drk-geek, 2020-06-17:
This will work on RPI 3B ?

lxne, 2020-06-17:
https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation#requirements

drk-geek, 2020-06-17:
Yes I saw that, I currently have 3B only

lxne, 2020-06-17:
Might work. I can't tell for sure.

drk-geek, 2020-06-18:
If there is any chance, I’ll try tonight and update you accordingly.

eric-h-st, 2020-07-15:
@drk-geek could you run the mirror from your console and see if you're getting any errors?
Instead of using the PM2, thus:
`pm2 stop mm
cd `~/MagicMirror
npm start run`

(Don't forget to turn it back on when you're done: "pm2 start mm")

drk-geek, 2020-07-16:
> @drk-geek could you run the mirror from your console and see if you're getting any errors?
> Instead of using the PM2, thus:
> `pm2 stop mm`
> `cd ~/MagicMirror`
> `npm start run`
> 
> (Don't forget to turn it back on when you're done: "pm2 start mm")

There is no error when npm start run , below is the log
pi@raspberrypi:~/MagicMirror $ npm start run

> magicmirror@2.12.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js "run"

[2020-07-16 08:23:27.325] [LOG]    Starting MagicMirror: v2.12.0
[2020-07-16 08:23:27.377] [LOG]    Loading config ...
[2020-07-16 08:23:27.397] [LOG]    Loading module helpers ...
[2020-07-16 08:23:27.403] [LOG]    No helper found for module: alert.
[2020-07-16 08:23:27.603] [LOG]    Initializing new module helper ...
[2020-07-16 08:23:27.607] [LOG]    Module helper loaded: updatenotification
[2020-07-16 08:23:27.618] [LOG]    No helper found for module: clock.
[2020-07-16 08:23:28.998] [LOG]    Initializing new module helper ...
[2020-07-16 08:23:28.999] [LOG]    Module helper loaded: calendar
[2020-07-16 08:23:29.010] [LOG]    No helper found for module: compliments.
[2020-07-16 08:23:29.027] [LOG]    No helper found for module: currentweather.
[2020-07-16 08:23:29.040] [LOG]    No helper found for module: weatherforecast.
[2020-07-16 08:23:30.310] [LOG]    Initializing new module helper ...
[2020-07-16 08:23:30.312] [LOG]    Module helper loaded: MMM-Screencast
[2020-07-16 08:23:30.318] [LOG]    No helper found for module: on-this-day.
[2020-07-16 08:23:30.326] [LOG]    Initializing new module helper ...
[2020-07-16 08:23:30.327] [LOG]    Module helper loaded: MMM-COVID19
[2020-07-16 08:23:30.458] [LOG]    Initializing new module helper ...
[2020-07-16 08:23:30.460] [LOG]    Module helper loaded: MMM-PIR-Sensor
[2020-07-16 08:23:30.548] [LOG]    Initializing new module helper ...
[2020-07-16 08:23:30.549] [LOG]    Module helper loaded: newsfeed
[2020-07-16 08:23:30.549] [LOG]    All module helpers loaded.
[2020-07-16 08:23:30.938] [LOG]    Starting server on port 8080 ... 
[2020-07-16 08:23:30.996] [LOG]    Server started ...
[2020-07-16 08:23:30.998] [LOG]    Connecting socket for: updatenotification
[2020-07-16 08:23:31.000] [LOG]    Connecting socket for: calendar
[2020-07-16 08:23:31.001] [LOG]    Starting node helper for: calendar
[2020-07-16 08:23:31.002] [LOG]    Connecting socket for: MMM-Screencast
[2020-07-16 08:23:31.004] [LOG]    Connecting socket for: MMM-COVID19
[2020-07-16 08:23:31.007] [LOG]    Starting node helper for: MMM-COVID19
[2020-07-16 08:23:31.011] [LOG]    Connecting socket for: MMM-PIR-Sensor
[2020-07-16 08:23:31.013] [LOG]    Connecting socket for: newsfeed
[2020-07-16 08:23:31.015] [LOG]    Starting node helper for: newsfeed
[2020-07-16 08:23:31.016] [LOG]    Sockets connected & modules started ...
[2020-07-16 08:23:31.569] [LOG]    Launching application.
[2020-07-16 08:23:36.635] [LOG]    Create new calendar fetcher for url: http://www.calendarlabs.com/ical-calendar/ics/102/Pakistan_Holidays.ics - Interval: 300000
[2020-07-16 08:23:36.827] [LOG]    Create new news fetcher for url: http://feeds.feedburner.com/com/YEor?format=xml - Interval: 300000
[2020-07-16 08:23:36.928] [INFO]   Checking git for module: MMM-Screencast
[2020-07-16 08:23:36.978] [INFO]   Checking git for module: on-this-day
[2020-07-16 08:23:37.021] [INFO]   Checking git for module: MMM-COVID19
[2020-07-16 08:23:37.076] [INFO]   Checking git for module: MMM-PIR-Sensor
[2020-07-16 08:23:38.210] [INFO]   Newsfeed-Fetcher: Broadcasting 50 items.
[2020-07-16 08:23:39.671] [INFO]   Calendar-Fetcher: Broadcasting 9 events.

eric-h-st, 2020-07-16:
Seems like the module is running,  but I cannot tell a lot more from the log, as no errors were reported.
The only thing I'd suggest is to check the hardware outside of the mirror. 
Google testing IR on raspberry pi. There're a few copy-paste examples out there that will allow you to test.

closed 2020-07-21 by eric-h-st