paviro/MMM-PIR-Sensor #101: Not working

Jagsi1981, 2020-07-15:
Hello, 
I am very new to raspberry pi and coding. I installed the PIR Sensor and I installed your MMM-PIR Sensor 
but it is not working - how can I try if the PIR Sensor is ok? 
which information I have to provide that somebode please can help me.

thanks

eric-h-st, 2020-07-15:
Did you install the `MMM-PIR` or `MMM-PIR-Sensor` project? These are two different projects.

Anyhow, please see this: [](https://thepihut.com/blogs/raspberry-pi-tutorials/raspberry-pi-gpio-sensing-motion-detection)

Jagsi1981, 2020-07-15:
I installed the MMM_PIR_Sensor project

Jagsi1981, 2020-07-15:
puh in the link I see a lot of coding - is this necessary when I am using the MMM-PIR-Sensor Module?

eric-h-st, 2020-07-15:
It's not. You wanted to test your hardware sensor, that's one way of doing it.
Another way would be to check if you see the USER_PRESENCE. 
You could try installing this: [https://github.com/eouia/MMM-NotificationTrigger](https://github.com/eouia/MMM-NotificationTrigger) that I use to display an alert upon motion, to see that your hardware is working for you. 

Suggested config: 
```` javascript
{
    module: "MMM-NotificationTrigger",
    config: {
        useWebhooks: false,
        triggers: [
        {
            trigger: "USER_PRESENCE",
            triggerSenderFilter: (sender) => {
                return sender.name == "MMM-PIR-Sensor"; 
            },
            triggerPayloadFilter: (payload) => {
                return  payload == true;
            },
            fires: [
            {
                fire: "SHOW_ALERT",
                payload: function() {
                    return {
                        type: "notification",        
                        title: "Hey There!",
                        message: "Welcome back!",
                        timer: 7500
                    }
                }
            }
            ]
        }
        ]
    }
}
````

closed 2020-07-21 by eric-h-st