paviro/MMM-PIR-Sensor #106: Try to update: Git pull && NPM install aborting because of package-lock.json

MadhunterNL, 2020-07-20:
When I try to git pull I get this error:

The following untracked working tree files would be overwritten by merge:
        package-lock.json
Please move or remove them before you merge.
Aborting

How can I avoid this error? I get this not only with this module but also with other module updates.

M1dn8, 2020-07-20:
cd MagicMirror/modules/MMM-PIR-Sensor
rm package-lock.json
git pull

this was my solution!

paviro, 2020-07-20:
I am not sure https://github.com/paviro/MMM-PIR-Sensor/blob/master/package-lock.json should be in the repo any insight @eric-h-st? Besides that I am unsure if remove the file is avoidable since it is generated on 'npm install' I think.

eric-h-st, 2020-07-20:
Please do a `git pull`. The said file has been removed.

closed 2020-07-20 by eric-h-st