paviro/MMM-PIR-Sensor #110: MM will not start after update 11/08/20

Tomxtal, 2020-08-12:
I have been using  MM-PIR-Sensor for the past few months with no problems but the last update looks like it has broken something.

I started the update by removing the older version with cd MagicMirror/modules rm -rf MMM-PIR-Sensor. Then installed the new version with  cd MagicMirror/modules git clone https://github.com/paviro/MMM-PIR-Sensor.git and npm install.

How I just get the white screen as if it is to start then it just goes to black screen with the cursor in the middle, if I disable MM-PIR-Sensor in config.js it starts just fine.

FritsdeGroot, 2020-08-12:
Hi I have a similar issues since the last update. This is what I get in my Pm2.log

> 020-08-11T21:30:51: PM2 log: ===============================================================================
> 2020-08-11T21:30:51: PM2 log: --- New PM2 Daemon started ----------------------------------------------------
> 2020-08-11T21:30:51: PM2 log: Time                 : Tue Aug 11 2020 21:30:51 GMT+0200 (Central European Summer Time)
> 2020-08-11T21:30:51: PM2 log: PM2 version          : 4.2.3
> 2020-08-11T21:30:51: PM2 log: Node.js version      : 10.22.0
> 2020-08-11T21:30:51: PM2 log: Current arch         : arm
> 2020-08-11T21:30:51: PM2 log: PM2 home             : /home/pi/.pm2
> 2020-08-11T21:30:51: PM2 log: PM2 PID file         : /home/pi/.pm2/pm2.pid
> 2020-08-11T21:30:51: PM2 log: RPC socket file      : /home/pi/.pm2/rpc.sock
> 2020-08-11T21:30:51: PM2 log: BUS socket file      : /home/pi/.pm2/pub.sock
> 2020-08-11T21:30:51: PM2 log: Application log path : /home/pi/.pm2/logs
> 2020-08-11T21:30:51: PM2 log: Worker Interval      : 30000
> 2020-08-11T21:30:51: PM2 log: Process dump file    : /home/pi/.pm2/dump.pm2
> 2020-08-11T21:30:51: PM2 log: Concurrent actions   : 2
> 2020-08-11T21:30:51: PM2 log: SIGTERM timeout      : 1600
> 2020-08-11T21:30:51: PM2 log: ===============================================================================
> 2020-08-11T21:30:52: PM2 log: [Watch] Start watching MagicMirror
> 2020-08-11T21:30:52: PM2 log: App [MagicMirror:0] starting in -fork mode-
> 2020-08-11T21:30:52: PM2 log: App [MagicMirror:0] online
> 2020-08-12T13:55:19: PM2 log: Change detected on path config/config.js for app MagicMirror - restarting
> 2020-08-12T13:55:19: PM2 log: Stopping app:MagicMirror id:0
> 2020-08-12T13:55:20: PM2 error: Error: kill EPERM
> 2020-08-12T13:55:20: PM2 error: Error: kill EPERM
> 2020-08-12T13:55:20: PM2 log: App [MagicMirror:0] exited with code [0] via signal [SIGINT]
> 2020-08-12T13:55:20: PM2 log: pid=835 msg=process killed
> 2020-08-12T13:55:20: PM2 log: App [MagicMirror:0] starting in -fork mode-
> 2020-08-12T13:55:20: PM2 log: App [MagicMirror:0] online
> 
If you want me to test something please let me know.

Kind regatds
Frits

eugenelai, 2020-08-16:
Having the same issue

brooklynrab, 2020-11-23:
Me too. Any solutions?

Dres49, 2021-01-24:
Dear all,

I encoutered the same problem and found the solution here:
https://forum.magicmirror.builders/topic/6991/mmm-pir-sensor-black-screen/7 

In short:
1) remove the module:
    cd ~/MagicMirror
    rm -rf modules/MMM-PIR-Sensor
2) reboot the raspi:
    sudo reboot
3) install the module again
    --> this worked for my installation.

Thanks to morozgrafix for his explanation - how to remove a module.

It seems that some updates cause changes in other configuration files and only a new installation of the
module (I had similar problem for other modules) sets the parameter back to working status.

Best regards,
Dres49.