paviro/MMM-PIR-Sensor #111: MM locked after turn On

subitus, 2020-09-11:
all is work fine after install but when the monitor turn on, mm is locked :

the clock is stopped. the other modules do not refresh

eric-h-st, 2020-09-11:
What do you have in the logs?

subitus, 2020-09-14:
[TAILING] Tailing last 15 lines for [mm] process (change the value with --lines option)
/home/pi/.pm2/logs/mm-error.log last 15 lines:
0|mm       |     at Module.require (internal/modules/cjs/loader.js:685:19)
0|mm       |     at require (internal/modules/cjs/helpers.js:16:16)
0|mm       |     at bindings (/home/pi/magicMirror/modules/MMM-PIR/node_modules/bindings/bindings.js:112:48)
0|mm       | [2020-09-11 17:14:18.869] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|mm       | [2020-09-11 17:14:18.869] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
0|mm       | npm ERR! code ELIFECYCLE
0|mm       | npm ERR! errno 1
0|mm       | npm ERR! magicmirror@2.12.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
0|mm       | npm ERR! Exit status 1
0|mm       | npm ERR! 
0|mm       | npm ERR! Failed at the magicmirror@2.12.0 start script.
0|mm       | npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
0|mm       | 
0|mm       | npm ERR! A complete log of this run can be found in:
0|mm       | npm ERR!     /home/pi/.npm/_logs/2020-09-11T15_15_06_393Z-debug.log

/home/pi/.pm2/logs/mm-out.log last 15 lines:
0|mm       | [2020-09-14 08:12:13.627] [LOG]    Connecting socket for: updatenotification
0|mm       | [2020-09-14 08:12:13.628] [LOG]    Connecting socket for: calendar
0|mm       | [2020-09-14 08:12:13.629] [LOG]    Starting node helper for: calendar
0|mm       | [2020-09-14 08:12:13.631] [LOG]    Connecting socket for: MMM-Todoist
0|mm       | [2020-09-14 08:12:13.632] [LOG]    Starting node helper for: MMM-Todoist
0|mm       | [2020-09-14 08:12:13.633] [LOG]    Connecting socket for: MMM-PIR-Sensor
0|mm       | [2020-09-14 08:12:13.634] [LOG]    Sockets connected & modules started ...
0|mm       | [2020-09-14 08:12:14.136] [LOG]    Launching application.
0|mm       | [2020-09-14 08:12:17.962] [LOG]    Create new calendar fetcher for url: http://localhost:8080/modules/calendars/***********.ics - Interval: 300000
0|mm       | [2020-09-14 08:12:17.971] [LOG]    Create new calendar fetcher for url: http://localhost:8080/modules/calendars/***********.ics - Interval: 300000
0|mm       | [2020-09-14 08:12:18.092] [INFO]   Checking git for module: MMM-AirQuality
0|mm       | [2020-09-14 08:12:18.136] [INFO]   Checking git for module: MMM-Todoist
0|mm       | [2020-09-14 08:12:18.164] [INFO]   Checking git for module: MMM-PIR-Sensor
0|mm       | [2020-09-14 08:12:18.339] [INFO]   Calendar-Fetcher: Broadcasting 6 events.
0|mm       | [2020-09-14 08:12:18.371] [INFO]   Calendar-Fetcher: Broadcasting 1 events.

---------------------------

after a restart, magicmirror works for 1 minute then it freezes

eric-h-st, 2020-09-14:
Please see previous thread about npm reinstall.
Seema like it's a module cache issue.

ChrizZz90, 2020-09-14:
I have the same issue, which thread about npm reinstall do you mean?

subitus, 2020-09-15:
I solved the problem by disabling one by one
I found the fault in the compliment module.
On compliments.js file
I wrote :

```
compliments: {
			anytime: [],
			morning: [],
			afternoon: [],
			evening: []
```

then adding “”

```
compliments: {
			anytime: [""],
			morning: [""],
			afternoon: [""],
			evening: [""]

```
the update interval was every 30 sec, that’s why it blocked.

closed 2020-09-15 by subitus

ChrizZz90, 2020-09-15:
was the solution for me, too. Should this issue been forwarded to the MagicMirror github section or is that an issue of this code?