paviro/MMM-PIR-Sensor #112: PIR not working on Magic Mirror

hec59, 2020-09-21:
I followed the instructions to the best of my ability and this does not work. I have a Pi4 4gb. If any one can write the instructions down step by step as if I was a 2 year old, this would be much appreciated. I have no experience in coding or linux. Thanks in advance

ginfts, 2020-09-23:
Same for me; I included the following within the config file, but the screen goes blank immediately and it seems impossible to re-awake it. I have had to go back to a backup .dmg

config: {
sensorPin: 22,
powerSaving: true,
powerSavingDelay: 30,

I'm using a fairly new Benq LCD screen, and the centre wire is connected to BCM 22.

TIA

brooklynrab, 2020-11-23:
Same for me. The screen goes black and blank as soon as I start up magic mirror, have to quit to get back to seeing anything on my screen. I remove the module from configure file and everything works again. Any ideas?

jptech484, 2021-01-22:
i have the same problem. i would love to know the solution.

mihed001, 2021-02-25:
I think it's a problem with rebuild of node-abi which makes MMM-PIR module crashing Magic Mirror.
Please try this command: **npm upgrade node-abi --depth=100**
This worked for me :-)

omrid594, 2021-03-01:
same issue here ->
```
	modules: [
		{
		module: "MMM-PIR-Sensor",
			config: {
				// default pin is 22 - view here https://pinout.xyz/
				powerSavingDelay: 60,
				preventHDMITimeout: 7,
				presenceIndicator: "fa-eye", // Customizing the indicator
				presenceOffIndicator: "fa-eye", // Customizing the indicator
				presenceIndicatorColor: "#f51d16", // Customizing the indicator
				presenceOffIndicatorColor: "#2b271c" // Customizing the indicator
			}
		},
```

and the output is 

```
	modules: [
		{
		module: "MMM-PIR-Sensor",
			config: {
				// default pin is 22 - view here https://pinout.xyz/
				powerSavingDelay: 60,
				preventHDMITimeout: 7,
				presenceIndicator: "fa-eye", // Customizing the indicator
				presenceOffIndicator: "fa-eye", // Customizing the indicator
				presenceIndicatorColor: "#f51d16", // Customizing the indicator
				presenceOffIndicatorColor: "#2b271c" // Customizing the indicator
			}
		},
```

Anyone?