paviro/MMM-PIR-Sensor #118: MagicMirror v2.15.0 breaks this module

john3300, 2021-04-01:
Here's the error log from startup:

```
[01.04.2021 09:43.57.629] [ERROR] ESC[31mWARNING! Could not load config file. Starting with default configuration. Error found: Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build/Release/epoll.node'ESC[39m
ESC[31mwas compiled against a different Node.js version usingESC[39m
ESC[31mNODE_MODULE_VERSION 73. This version of Node.js requiresESC[39m
ESC[31mNODE_MODULE_VERSION 85. Please try re-compiling or re-installingESC[39m
ESC[31mthe module (for instance, using `npm rebuild` or `npm install`).ESC[39m
[01.04.2021 09:43.57.668] [ERROR] App threw an error during load
[01.04.2021 09:43.57.672] [ERROR] Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build/Release/epoll.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 73. This version of Node.js requires
NODE_MODULE_VERSION 85. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (electron/js2c/asar_bundle.js:5:1812)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1203:18)
    at Object.func [as .node] (electron/js2c/asar_bundle.js:5:1812)
    at Module.load (internal/modules/cjs/loader.js:992:32)
    at Module._load (internal/modules/cjs/loader.js:885:14)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12738)
    at Module.require (internal/modules/cjs/loader.js:1032:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at bindings (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bindings/bindings.js:112:48)
    at /home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/epoll.js:7:31
[01.04.2021 09:43.57.674] [ERROR] Whoops! There was an uncaught exception...
[01.04.2021 09:43.57.681] [ERROR] Error: The module '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build/Release/epoll.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 73. This version of Node.js requires
NODE_MODULE_VERSION 85. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (electron/js2c/asar_bundle.js:5:1812)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1203:18)
    at Object.func [as .node] (electron/js2c/asar_bundle.js:5:1812)
    at Module.load (internal/modules/cjs/loader.js:992:32)
    at Module._load (internal/modules/cjs/loader.js:885:14)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12738)
    at Module.require (internal/modules/cjs/loader.js:1032:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at bindings (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/bindings/bindings.js:112:48)
    at /home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/epoll.js:7:31

```

mrouillard, 2021-04-01:
I believe this could be solved by adding a post-install script to rebuild electron, as described in [this thread](https://forum.magicmirror.builders/topic/12471/black-screen-with-mmm-pir-sensor-after-update/4)
So to sum up: go to the modules/MMM-PIR-Sensor, then `npm install electron-rebuild`
then `npm install`

FritsdeGroot, 2021-04-20:
Hi,

Same issue here. I did a reinstall of the module and the instructions above  

output of **npm install electron-rebuild**:
```
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm notice created a lockfile as package-lock.json. You should commit this file.
+ electron-rebuild@1.11.0
removed 1 package, updated 1 package and audited 164 packages in 19.9s
found 0 vulnerabilities
```

output of **npm install**:
```
> epoll@4.0.0 install /home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll
> node-gyp rebuild

gyp ERR! UNCAUGHT EXCEPTION
gyp ERR! stack Error: Cannot find module '../dotjs/validate'
gyp ERR! stack     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
gyp ERR! stack     at Function.Module._load (internal/modules/cjs/loader.js:562:25)
gyp ERR! stack     at Module.require (internal/modules/cjs/loader.js:692:17)
gyp ERR! stack     at require (internal/modules/cjs/helpers.js:25:18)
gyp ERR! stack     at Object.<anonymous> (/usr/lib/node_modules/npm/node_modules/har-validator/node_modules/ajv/lib/compile/index.js:8:25)
gyp ERR! stack     at Module._compile (internal/modules/cjs/loader.js:778:30)
gyp ERR! stack     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
gyp ERR! stack     at Module.load (internal/modules/cjs/loader.js:653:32)
gyp ERR! stack     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
gyp ERR! stack     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
gyp ERR! System Linux 5.10.17-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll
gyp ERR! node -v v10.24.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! This is a bug in `node-gyp`.
gyp ERR! Try to update node-gyp and file an Issue if it does not help:
gyp ERR!     <https://github.com/nodejs/node-gyp/issues>
npm ERR! code ELIFECYCLE
npm ERR! errno 7
npm ERR! epoll@4.0.0 install: `node-gyp rebuild`
npm ERR! Exit status 7
npm ERR!
npm ERR! Failed at the epoll@4.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-04-20T12_19_49_643Z-debug.log
```

DanielHabenicht, 2021-10-17:
For me a simple
```bash
npm install
npm rebuild
```
Fixed the problem.