paviro/MMM-PIR-Sensor #123: Execute 'sudo chmod u+s /opt/vc/bin/tvservice' does not work in Bullseye

aegis1980, 2021-11-14:
In latest Raspberry Pi OS release (Bullseye) the following command from Installation instructions in `readme.md` does not work:

4. Execute `sudo chmod u+s /opt/vc/bin/tvservice && sudo chmod u+s /bin/chvt` to allow turning on/off the hdmi output.

```
chmod: cannot access '/opt/vc/bin/tvservice': No such file or directory
```

`chvt` changes do execute.

skippy-oz, 2021-11-16:
I think may found something when running tvservice I get the below error

tvservice is not supported when using the vc4-kms-v3d driver.
Similar features are available with standard linux tools
such as modetest from libdrm-tests.

2wenty2wo, 2021-11-28:
I have the exact same issue, "no such file or directory".
Is there a fix/work-around for this?

Thank you.

skippy-oz, 2021-11-28:
I downgraded to the latest version of buster I had an image for as Bulleye change the video driver and tvservice is not supported

2wenty2wo, 2021-12-06:
I was really hoping I wouldn't have to do that. I only just finished setting up Bullseye.

Gillemo, 2021-12-19:
this module was created on buster, the latest version is bullseye. 
In the buster version the tvservice file is located as shown: /opt/vc/bin/tvservice
however, on the bullseye version this file is here: /bin/tvservice

Thats why i can't find it. 
you can recreate the same path and copy the file to "fool" the os. 

`cd /`
`cd opt`
`sudo mkdir vc`
`cd vc`
`sudo mkdir bin`
`sudo cp /bin/tvservice /opt/vc/bin/'

Now is you should have created the same path.

mike4192, 2021-12-27:
I believe `tvservice` is no longer used by this module, and the instructions are outdated. If anything, the instructions I think should be replaced with:
`Execute `sudo chmod u+s /usr/bin/vcgencmd && sudo chmod u+s /bin/chvt`

I opened a PR to address this (https://github.com/paviro/MMM-PIR-Sensor/pull/126)