paviro/MMM-PIR-Sensor #125: Module only works once - doesn't go back to power saving

lowanger, 2021-11-20:
Hello everyone,

i just updated my MM to 2.17 and also my modules. Everything works fine (finally) except the MMM-PIR-Sensor.

The module did detect my first motion turned on and after the set amount of time (60s) turned the screen off. The issue is, that after moving again the module does deactivate the PowerSaving anymore…
HOWEVER after some time of the MM running it did detect the motion again… Now the opossite issue happens: It is not turning “off” anymore after the 60s…

Anyone got an idea on what to do?

lowanger, 2021-11-20:
Oh and the settings in use are:

{
module: 'MMM-PIR-Sensor',
config: {
sensorPIN: 17,
powerSaving: true,
powerSavingDelay: 60,
}

Gillemo, 2021-12-19:
Same issue :( hoping for a fix.
I get the screen back on with ssh 
`tvservice -p`
but not with motion, i'm stuck for a week already!

abauer361, 2022-01-03:
Receiving the same problem here as well!