paviro/MMM-PIR-Sensor #128: Error: Could not detect abi for version 16.0.6

kikig1, 2022-01-04:
Hi,
i've tried it sometimes with a fully new instllation of raspi and MagicMirror.
i've tried npm install electron-rebuild

npm install in the MMM-PIR-Sensor always stops with
`An unhandled error occurred inside electron-rebuild
Could not detect abi for version 16.0.6 and runtime electron.  Updating "node-abi" might help solve this issue if it is a new release of electron

Error: Could not detect abi for version 16.0.6 and runtime electron.  Updating "node-abi" might help solve this issue if it is a new release of electron
    at Object.getAbi (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/node-abi/index.js:36:9)
    at new Rebuilder (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/rebuild.js:126:48)
    at rebuildWithOptions (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/rebuild.js:404:23)
    at Object.doRebuild [as rebuild] (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/rebuild.js:427:16)
    at /home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/cli.js:124:33
    at Generator.next (<anonymous>)
    at fulfilled (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild/lib/src/cli.js:6:58)
npm ERR! code 255
`

Are there any solutions fpr this?
Thanks for any tips ;)
best regrads

0synonym, 2022-01-05:
I had the same issue. I found this interim solution / workaround [electron-rebuild and MagicMirror v2.18 (and more) ](https://forum.magicmirror.builders/topic/16074/electron-rebuild-and-magicmirror-v2-18-and-more) by.
Unfortunately any other rebuild failed.

bugsounet, 2022-01-10:
Hi,

I have open a [PR](https://github.com/paviro/MMM-PIR-Sensor/pull/130) for correct it

@bugsounet