paviro/MMM-PIR-Sensor #129: postinstall issue [LIBERTY LIBERY LIBERTY]

Coumodo, 2022-01-10:
clean fresh rasbian and MagicMirror installation.
Everthing up to date and fine.
I get an strange issue trying to install MMM-PIR-Module:
[image]
 these wall of symbols runs forever without stoping or change something in the module folder...

already deleted the MMM-PIR-Sensor folder and git hard reset to start again clone git directory - same weird result.

bugsounet, 2022-01-10:
Hi @Coumodo,
I have open a [PR](https://github.com/paviro/MMM-PIR-Sensor/pull/130) for correct it

You can follow this [link](https://forum.magicmirror.builders/topic/16074/electron-rebuild-and-magicmirror-v2-18-and-more) for correct it

@bugsounet

closed 2022-01-10 by Coumodo

cyber152, 2022-01-11:
https://www.msn.com/fr-fr/actualite/other/un-d%C3%A9veloppeur-sabote-son-projet-open-source-et-paralyse-des-milliers-d-applications/ar-AASCKtj?ocid=msedgntp

https://www.theverge.com/2022/1/9/22874949/developer-corrupts-open-source-libraries-projects-affected