paviro/MMM-PIR-Sensor #131: install problems

shgtjgh, 2022-01-23:
An unhandled error occurred inside electron-rebuild

during npm install
any suggestions?

Jan-2005, 2022-01-27:
dieser Fehler ?`An unhandled error occurred inside electron-rebuild
Could not detect abi for version 16.0.7 and runtime electron.  Updating "node-ab                            i" might help solve this issue if it is a new release of electron

Error: Could not detect abi for version 16.0.7 and runtime electron.  Updating "                            node-abi" might help solve this issue if it is a new release of electron
    at Object.getAbi (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/n                            ode-abi/index.js:36:9)
    at new Rebuilder (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/e                            lectron-rebuild/lib/src/rebuild.js:126:48)
    at rebuildWithOptions (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modu                            les/electron-rebuild/lib/src/rebuild.js:404:23)
    at Object.doRebuild [as rebuild] (/home/pi/MagicMirror/modules/MMM-PIR-Senso                            r/node_modules/electron-rebuild/lib/src/rebuild.js:427:16)
    at /home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/electron-rebuild                            /lib/src/cli.js:124:33
    at Generator.next (<anonymous>)
    at fulfilled (/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/elect                            ron-rebuild/lib/src/cli.js:6:58)
npm ERR! code 255
npm ERR! path /home/pi/MagicMirror/modules/MMM-PIR-Sensor
npm ERR! command failed
npm ERR! command sh -c node_modules/.bin/electron-rebuild -e ../../node_modules/                            electron

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-01-27T20_04_41_227Z-debug-0.log
`

thjansen72, 2022-02-05:
This worked for me:
https://forum.magicmirror.builders/topic/16074/electron-rebuild-and-magicmirror-v2-18-and-more